Cloud-Automation/node-modbus #164: Long Term Tests

stefanpoeter, 2018-01-13:
I want to setup a client and server machine to monitor jsModbus on the long run and monitor cpu and memory consumption. I happen to have some good experiences with munin. 

Does anyone have any suggestions how this setup should look like? I am thinking of two simple virtual machines, one as a client and one as a server. The client is executing every available function code on the server.

Thoughts and Ideas are welcome @psorowka, @AnthonyMujic

Nedgeva, 2018-01-21:
@stefanpoeter very good idea!

> I am thinking of two simple virtual machines
Even better to isolate client and server as docker images, this would allow to easily reproduce tests for everyone.

I think in server simulation scenario it should emit predefined  connection errors (just like timeouts and unexpected socket closings or throwing incorrect payloads), because sometimes mb client implementations starts uncontrolled memory consumption after handling these errors.

stefanpoeter, 2018-01-22:
@Nedgeva good ones!

Nedgeva, 2018-01-22:
@stefanpoeter thanks!

Speaking of sockets - i think there are at least two ways of establishing connection between server and client. One of is to reuse existing connection where polling happens, another is to create new connection each time client tries to request data from mbus server. It's not common nowadays, however some good ol' PLCs has been packed with this kind of mbus servers, where dropping connection after response was usual behavior.

What do you think? Which one is preferable?

stefanpoeter, 2018-01-22:
I am imagening the following scenario:

Server and Clients are on the same machine.

Client 1 is opening up one connection to the client and makes every
available requests in a loop.
Client 2 is doing the same but reconnecting after each request group.
Client 3 is doing what client 1 does, but randomly kills the whole process.
Client 4 is fireing weird data to the client periodically.

I think we can do all of them on one virtual machine.

15:09 Uhr:

-- 
Dipl.-Inf. Stefan Pöter

*CLOUD AUTOMATION*
Connecting Technologies

Katernberger Straße 107
45327 Essen

NRW, Germany

*phone* // +49 (0) 201 759 396 94
*e-mail* // stefan.poeter@cloud-automation.de

Wichtiger Hinweis: Diese Information ist für den Gebrauch durch die Person
oder die Firma/Organisation bestimmt, die in der Empfängeradresse benannt
ist. Wenn Sie nicht der angegebene Empfänger sind, nehmen Sie bitte zur
Kenntnis, dass Weitergabe, Kopieren, Verteilung oder Nutzung des Inhalts
dieser E-Mail-Übertragung unzulässig sind. Falls Sie diese E-Mail
irrtümlich erhalten haben, benachrichtigen Sie den Absender bitte
unverzüglich telefonisch oder durch eine E-Mail.

„This message is intended only for the named recipient. If you are not the
intended recipient you are notified that disclosing, copying, distributing
or taking any action in reliance on the contents of this information is
strictly prohibited.”

AnthonyMujic, 2018-01-25:
> *phone* // +49 (0) 201 759 396 94 <+49%20201%2075939694>

They all sound like good ideas to me. I have no practical experience on how
to implement such a system, but I'll be interested to see how it is done.

stefanpoeter, 2018-03-27:
I've setup a munin server on jsmodbus.cloud-automation.de. You can see the process memory consumption and the cpu usage under the category jsmodbus.

closed 2018-04-30 by stefanpoeter