Cloud-Automation/node-modbus #168: v3 Dev State

stefanpoeter, 2018-01-26:
The way I see it, we are missing some fcs on the TCP Server Response Handler. After that we can move to a beta phase and start testing v3. What do you think @AnthonyMujic ?

AnthonyMujic, 2018-01-27:
> to @AnthonyMujic <https://github.com/anthonymujic>.

I think that is a great idea. There are fc's 15 and 16 then we can test to
work out what more is required.

AnthonyMujic, 2018-02-11:
Fc's 15 and 16 are complete.
@stefanpoeter, let me know if there are any issues, or if there is anything else in v3 you would like me to work on.

stefanpoeter, 2018-02-11:
@AnthonyMujic great work, I would say we can move soon to a beta version and create some scripts for long running tests.

kevingarman, 2018-02-12:
@stefanpoeter, probably not a great place to post this, but just wanted to say thanks for this project.  I've been using it a bit and it's been solid for me.  I'm testing v3 now because I need the #149 functionality...looking forward to the v3 release!

stefanpoeter, 2018-02-18:
@kevingarman How are you tests going?

kevingarman, 2018-02-20:
@stefanpoeter So far so good!

closed 2018-04-30 by stefanpoeter