Cloud-Automation/node-modbus #171: Index Out of Range Error onfirst Call

mariam-crissi, 2018-02-07:
I am using the following code.
       
```
readingFunction: function(req,res) {
    
          console.log("Inside Modbus Connection");
          console.log(req);
          var modbushost = req.modbushost;
          var modbusport = req.modbusport;
          var modbusunitid = req.modbusunitid;
          var modbusunitfrom = req.modbusunitfrom;
          var modbusunitcount = req.modbusunitcount;
          console.log(modbushost);
          console.log(modbusport);
          console.log(modbusunitid);
          console.log(modbusunitfrom);
          console.log(modbusunitcount);
          // create a modbus client
         var client = modbus.client.tcp.complete({ 
               'host'              : modbushost, 
               'port'              : modbusport,
               'autoReconnect'     : true,
               'reconnectTimeout'  : 1000,
               'timeout'           : 5000,
               'unitId'            : modbusunitid
       });

      console.log("client Info");
      console.log(client);
      client.connect();
      client.on('connect', function () {
          client.readHoldingRegisters(modbusunitfrom, modbusunitcount).then(function (resp) {
                 console.log(resp); 
                 res({
                     success: true,
                     response: resp
                 });
         })
         .catch(function (error) {
                      console.log(error);
                      res({
                            success: false
                      })
                      client.close()
         })
         .finally(function () {
                   client.close()
         })
        
    });

   client.on('error', function (err) {
         console.log(err); 
   })
  } 

```
On the first call to this function, it returns the following error.

```
buffer.js:1024
    throw new errors.RangeError('ERR_INDEX_OUT_OF_RANGE');
    ^

RangeError [ERR_INDEX_OUT_OF_RANGE]: Index out of range
    at checkOffset (buffer.js:1024:11)
    at Buffer.readUInt8 (buffer.js:1064:5)
    at handleErrorPDU (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-client-core.js:76:27)
    at Object.<anonymous> (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-client-core.js:117:11)
    at Object.emit (/home/maria/Desktop/Modbus-Final/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.<anonymous> (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-tcp-client.js:128:14)
    at Socket.emit (events.js:159:13)
    at addChunk (_stream_readable.js:265:12)
    at readableAddChunk (_stream_readable.js:252:11)
    at Socket.Readable.push (_stream_readable.js:209:10)
    at TCP.onread (net.js:608:20)

```
On the second call , it returns the required output.

I have tried a lot to handle this error that popped out from the line  ```client.readHoldingRegisters(modbusunitfrom, modbusunitcount).then(function (resp)`

How can I handle this issue within the code?

stefanpoeter, 2018-02-07:
what are the values from modbusunitfrom and modbusunitcount? can you log them?

mariam-crissi, 2018-02-07:
@stefanpoeter modbusunitfrom:600 modbusunitcount:modbusregs*2; ,where modbusregs =10..I am facing this issue on the first call..If I try to call the function with same input later,it will return the output correctly.

mariam-crissi, 2018-02-07:
```
DEBUG : init connect
DEBUG : connect connect
DEBUG : Starting read holding registers request.
DEBUG : connect ready
DEBUG : Trying to flush data.
DEBUG : ready waiting
DEBUG : Sending pdu to the socket.
DEBUG : Data flushed.
DEBUG : received data
DEBUG : received data
DEBUG : PDU extracted
DEBUG : received data
buffer.js:1024
    throw new errors.RangeError('ERR_INDEX_OUT_OF_RANGE');
    ^

RangeError [ERR_INDEX_OUT_OF_RANGE]: Index out of range
    at checkOffset (buffer.js:1024:11)
    at Buffer.readUInt8 (buffer.js:1064:5)
    at handleErrorPDU (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-client-core.js:76:27)
    at Object.<anonymous> (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-client-core.js:117:11)
    at Object.emit (/home/maria/Desktop/Modbus-Final/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.<anonymous> (/home/maria/Desktop/Modbus-Final/node_modules/jsmodbus/src/modbus-tcp-client.js:128:14)
    at Socket.emit (events.js:159:13)
    at addChunk (_stream_readable.js:265:12)
    at readableAddChunk (_stream_readable.js:252:11)
    at Socket.Readable.push (_stream_readable.js:209:10)
    at TCP.onread (net.js:608:20)
```

stefanpoeter, 2018-02-07:
Seems like there is a problem with the response from your server. Can you look at the response in wireshark? Seems like it is sending to much or wrong data.

mariam-crissi, 2018-02-08:
@stefanpoeter I got the output correctly on second call..I can't able to handle this issue in my code...Any solution to handle this issue.

stefanpoeter, 2018-02-08:
As I said, can you monitor the response from your modbus server and post it here?

mariam-crissi, 2018-02-09:
[image]

mariam-crissi, 2018-02-09:
@stefanpoeter  I have added...Please help me to resolve the error
```
buffer.js:1024
    throw new errors.RangeError('ERR_INDEX_OUT_OF_RANGE');
    ^

RangeError [ERR_INDEX_OUT_OF_RANGE]: Index out of range
    at checkOffset (buffer.js:1024:11)
    at Buffer.readUInt8 (buffer.js:1064:5)
    at handleErrorPDU (/home/maria/Desktop/modbusNEW/node_modules/jsmodbus/src/modbus-client-core.js:76:27)
    at Object.<anonymous> (/home/maria/Desktop/modbusNEW/node_modules/jsmodbus/src/modbus-client-core.js:117:11)
    at Object.emit (/home/maria/Desktop/modbusNEW/node_modules/stampit-event-bus/src/stampit-event-bus.js:20:38)
    at Object.<anonymous> (/home/maria/Desktop/modbusNEW/node_modules/jsmodbus/src/modbus-tcp-client.js:128:14)
    at Socket.emit (events.js:159:13)
    at addChunk (_stream_readable.js:265:12)
    at readableAddChunk (_stream_readable.js:252:11)
    at Socket.Readable.push (_stream_readable.js:209:10)
    at TCP.onread (net.js:608:20)

```

stefanpoeter, 2018-02-09:
From the WireShark Log I can say that your request has the transaction id 1 and the response has the transaction id 0. Thats wrong, the server should answer with the same transaction id as the request. What server are you using?

stefanpoeter, 2018-02-21:
I am closing this due to no response. If you got news we can reopen the issue.

closed 2018-02-21 by stefanpoeter