Cloud-Automation/node-modbus #175: ECONNRESET

ortizvinicius, 2018-02-19:
Hello, i am getting this error sporadically when writing single coils:

´´´
error:  Error: read ECONNRESET
    at exports._errnoException (util.js:1020:11)
    at TCP.onread (net.js:568:26)
´´´

I connect the client once in the top of script, and call the write method along the day, so i think the problem is the socket closing before i try to write again.

What can be done to prevent this to happen?

stefanpoeter, 2018-02-21:
Can you present some code to reproduce the problem?

ortizvinicius, 2018-02-23:
Yes, something like this (shorten version):

```
const motorsData = require('../data/motors'),
      config = require('../data/config');

var modbus = false,
    modbusClient,
    modbusClientAddress,
    modbusClientPort;

config.getProtocols()
  .then(function getProtocolsSucess(protocols){
    if(protocols.modbus){
      modbus = require('jsmodbus');
      modbusClient = modbus.client.tcp.complete({
        'host': protocols.modbus.clientAddress,
        'port': protocols.modbus.clientPort,
        'unitId': 1,
        'logEnabled': false,
        'logLevel': 'debug' 
      });

      modbusClient.on('connect', function modbusConnect(){
        winston.log('info', 'Ok');
      });

      modbusClient.on('error', (e) => {
        winston.log('error', 'Exception');
        winston.log('error', e);
      });
      
      modbusClient.connect();
    }
  })
  .catch(function getProtocolsError(e){ //Here is where i get the exception, when i try to write the single coil
    someFunction here....then(() => {
        winston.log('error', ' Exception');
        winston.log('error', e);
      })
      .catch((e) => {
        winston.log('error', 'Exception');
        winston.log('error', e);
      });
  });

/**
 * @param {Boolean} status
 * @param {Number} silo
 * @param {Number} controller
 * @param {Number} motorId
 */
exports.motorAction = (status, silo, controller, motorId) => {
  return new Promise((successCallback, errorCallback) => {
    try {
        let motorParameters;

        some Functions here...
        .then(() => {

          function turnON(){
           
            if(modbus){ //Controle via Modbus
           
              modbusClient.writeSingleCoil(motorParameters.modbusControl, true).then(function modbusWrite(resp){
                
              }, function modbusWriteError(e){
                some Functions here....then(() => { handleError(e); })
                  .catch((e) => {
                    winston.log('error', 'Exception');
                    winston.log('error', e);
                  });
              });
            }
          }
    } catch (e) { 
      handleError(e);
    }
  });
};
```

ortizvinicius, 2018-02-23:
Do you think this could solve the problem?:

```
client.on('close', function () {
  console.log('Client closed, stopping interval.')

  if (!closedOnPurpose) {
    client.reconnect()
  }
})
```

Found in https://github.com/Cloud-Automation/node-modbus/blob/master/examples/tcp/Reconnect.js

stefanpoeter, 2018-02-23:
This is a lot of code and you also export some functions, can't tell what is actually going on in your application. The error can happen anywhere. Try to reproduce the problem with a short script with only the modbus calls.

limengsong, 2018-05-20:
excuse， please show the application how to use？
now I download the code. but i don't know how to do next?
I install some package  just like jsmodbus ,mocha and sinon

closed 2018-06-30 by stefanpoeter