Cloud-Automation/node-modbus #176: V3 - F6 : write single register [bug fixed]

profelec, 2018-02-21:
Hi,
for the example to work, I made that :
/src/tcp-server-response-handler.js
ligne 162 :
	```
} else {
+ 		this._server.emit('preWriteSingleRegister', request.body.address, this._server.holding) 
        	this._server.holding.writeUInt16BE(responseBody.value, responseBody.address * 2)
+       	this._server.emit('WriteSingleRegister', responseBody.address, this._server.holding) 
      	}
```

/examples/tcp/SimpleServer.js
ligne 27 :
```
server.on('preWriteSingleRegister', function (value, address) {
	console.log('Write Single Register')
-  	console.log('Original {register, value}: {', address, ',', server.holding.readUInt16BE(address), '}’)
+ 	console.log('Original {register, value}: {', address, ',', server.holding.readUInt16BE(address*2), '}')
})

server.on('WriteSingleRegister', function (value, address) {
-  	console.log('New {register, value}: {', address, ',', server.holding.readUInt16BE(address), '}’)
+  	console.log('New {register, value}: {', address, ',', server.holding.readUInt16BE(address*2), '}')
})

```

stefanpoeter, 2018-02-21:
I am not sure these events are necessary since you can just overwrite the default request handler. What do you think?

profelec, 2018-02-21:
I do not see how to do that.
This was for the example to work with SimpleServer.js

profelec, 2018-02-22:
functions 15 and 16 work with events
it's the same way

stefanpoeter, 2018-03-06:
I do not intend to use every feature from the v2 version. Do you actually use these events or are those just messing with the example (which is still ment to be for v2)?

profelec, 2018-03-08:
I am switching from version 1 to version 3. 
I am currently using functions 3 and 6 and events.

stefanpoeter, 2018-03-08:
Well it can't hurt to keep these events in place. Do want to prepare a pull request for those (and the other) events?

profelec, 2018-03-08:
ok

closed 2018-04-30 by stefanpoeter