Cloud-Automation/node-modbus #177: V3 - F3 : read holding registers [bug fixed]

profelec, 2018-02-21:
Hi,
for the example to work, I made that :
/src/tcp-server-response-handler.js
ligne 63 :
`
     let ReadHoldingRegistersResponseBody = require('./response/read-holding-registers.js')
-   let responseBody = ReadHoldingRegistersResponseBody.fromRequest(request.body, this._server.holdingRegisters)
+  let responseBody = ReadHoldingRegistersResponseBody.fromRequest(request.body, this._server.holding)
 +  this._server.emit('readHoldingRegisters', responseBody.values)
      let response = ModbusTCPResponse.fromRequest(request, responseBody)
      let payload = response.createPayload()
      cb(payload)
`

/examples/tcp/SimpleServer.js
`
 server.on('readHoldingRegisters', function (values) { 
   /* Implement your own */
   console.log('readHoldingRegisters : ', values)  
 })
`

stefanpoeter, 2018-02-21:
Related to the emit call.see my.xomment on issue #176.

profelec, 2018-02-22:
there is still a typo to `this._server.holdingRegisters` ;-)

closed 2018-04-30 by stefanpoeter