Cloud-Automation/node-modbus #184: V3 Server - F1: Read Coils

RenanGarcia, 2018-03-01:
Hi, I found the following problem with reading the coils with the server code.
Notice that the buffer values did not change during the execution of the tests, as my application's logs show in connection event.
As you can see, the server is returning unexpected values in some situations:

- In request 1 and 2, the response values are the same, even with requests for different reference addresses

- In request 3 and 4, when the number of bits is less than the allocated, the response values are wrong, regardless of the reference number indicated in the request

```
Initialization values:
##########################################################
coils: Buffer.alloc( 1 , 0)
discrete: Buffer.alloc( 1 , 0)
server.coils: <Buffer 05>
server.discrete: <Buffer 02>
```
Request 1: F1, **Reference number: 0, Bit Count: 8**
```
Expected: [ 1, 0, 1, 0, 0, 0, 0, 0 ]
Response: [ 1, 0, 1, 0, 0, 0, 0, 0 ]
```
logs
```
  modbus tcp server new connection coming in +0ms
  server.on('connection'... values in memory:
  server.on('connection'... server.coils:  <Buffer 05>
  server.on('connection'... server.discrete: <Buffer 02>
  modbus tcp client socket new data coming in +3ms
  tcp-server-request-handler this._buffer <Buffer 00 01 00 00 00 06 01 01 00 00 00 08> +0ms
  tcp-request tcp header complete, id 1 protocol 0 length 6 unitId 1 +1ms
  tcp-request buffer <Buffer 00 01 00 00 00 06 01 01 00 00 00 08> +0ms
  tcp-server-request-handler request ModbusTCPRequest {
  _id: 1,
  _protocol: 0,
  _length: 6,
  _unitId: 1,
  _body: ReadCoilsRequestBody { _fc: 1, _start: 0, _count: 8 } } +1ms
  tcp-request no enough data in the buffer yet +2ms
  tcp-server-request-handler request null +0ms
  read-coils-response response ReadCoilsResponseBody {
  _fc: 1,
  _coils: <Buffer 05>,
  _numberOfBytes: 1,
  _valuesAsBuffer: <Buffer 05>,
  _valuesAsArray: [ 1, 0, 1, 0, 0, 0, 0, 0 ] } +2ms
  modbus tcp client socket no request to process +1ms
  modbus tcp client socket response flushed +0ms
```
Request 2: F1, **Reference number: 1, Bit Count: 7**
```
Expected: [ 0, 1, 0, 0, 0, 0, 0 ]
Response: [ 1, 0, 1, 0, 0, 0, 0, 0 ]
```
logs
```
  modbus tcp server new connection coming in +11m
  server.on('connection'... values in memory:
  server.on('connection'... server.coils:  <Buffer 05>
  server.on('connection'... server.discrete: <Buffer 02>
  modbus tcp client socket new data coming in +1ms
  tcp-server-request-handler this._buffer <Buffer 00 01 00 00 00 06 01 01 00 01 00 07> +0ms
  tcp-request tcp header complete, id 1 protocol 0 length 6 unitId 1 +0ms
  tcp-request buffer <Buffer 00 01 00 00 00 06 01 01 00 01 00 07> +0ms
  tcp-server-request-handler request ModbusTCPRequest {
  _id: 1,
  _protocol: 0,
  _length: 6,
  _unitId: 1,
  _body: ReadCoilsRequestBody { _fc: 1, _start: 1, _count: 7 } } +0ms
  tcp-request no enough data in the buffer yet +1ms
  tcp-server-request-handler request null +0ms
  read-coils-response response ReadCoilsResponseBody {
  _fc: 1,
  _coils: <Buffer 05>,
  _numberOfBytes: 1,
  _valuesAsBuffer: <Buffer 05>,
  _valuesAsArray: [ 1, 0, 1, 0, 0, 0, 0, 0 ] } +0ms
  modbus tcp client socket no request to process +0ms
  modbus tcp client socket response flushed +0ms
```
Request 3: F1, **Reference number: 0, Bit Count: 4**
```
Expected: [ 1, 0, 1, 0, ]
Response: [ 0, 0, 0, 0, 0, 0, 0, 0]
```
logs
```
  modbus tcp server new connection coming in +21s
  server.on('connection'... values in memory:
  server.on('connection'... server.coils:  <Buffer 05>
  server.on('connection'... server.discrete: <Buffer 02>
  modbus tcp client socket new data coming in +1ms
  tcp-server-request-handler this._buffer <Buffer 00 01 00 00 00 06 01 01 00 00 00 04> +0ms
  tcp-request tcp header complete, id 1 protocol 0 length 6 unitId 1 +0ms
  tcp-request buffer <Buffer 00 01 00 00 00 06 01 01 00 00 00 04> +0ms
  tcp-server-request-handler request ModbusTCPRequest {
  _id: 1,
  _protocol: 0,
  _length: 6,
  _unitId: 1,
  _body: ReadCoilsRequestBody { _fc: 1, _start: 0, _count: 4 } } +0ms
  tcp-request no enough data in the buffer yet +0ms
  tcp-server-request-handler request null +1ms
  read-coils-response response ReadCoilsResponseBody {
  _fc: 1,
  _coils: <Buffer 00>,
  _numberOfBytes: 1,
  _valuesAsBuffer: <Buffer 00>,
  _valuesAsArray: [ 0, 0, 0, 0, 0, 0, 0, 0 ] } +0ms
  modbus tcp client socket no request to process +0ms
  modbus tcp client socket response flushed +0ms
```
Request 4: F1, **Reference number: 1, Bit Count: 4**
```
Expected: [ 0, 1, 0, 0 ]
Response: [ 0, 0, 0, 0, 0, 0, 0, 0 ]
```
logs
```
  modbus tcp server new connection coming in +21s
  server.on('connection'... values in memory:
  server.on('connection'... server.coils:  <Buffer 05>
  server.on('connection'... server.discrete: <Buffer 02>
  modbus tcp client socket new data coming in +1ms
  tcp-server-request-handler this._buffer <Buffer 00 01 00 00 00 06 01 01 00 01 00 04> +0ms
  tcp-request tcp header complete, id 1 protocol 0 length 6 unitId 1 +0ms
  tcp-request buffer <Buffer 00 01 00 00 00 06 01 01 00 01 00 04> +0ms
  tcp-server-request-handler request ModbusTCPRequest {
  _id: 1,
  _protocol: 0,
  _length: 6,
  _unitId: 1,
  _body: ReadCoilsRequestBody { _fc: 1, _start: 1, _count: 4 } } +0ms
  tcp-request no enough data in the buffer yet +0ms
  tcp-server-request-handler request null +0ms
  read-coils-response response ReadCoilsResponseBody {
  _fc: 1,
  _coils: <Buffer 00>,
  _numberOfBytes: 1,
  _valuesAsBuffer: <Buffer 00>,
  _valuesAsArray: [ 0, 0, 0, 0, 0, 0, 0, 0 ] } +0ms
  modbus tcp client socket no request to process +1ms
  modbus tcp client socket response flushed +0ms
```

stefanpoeter, 2018-03-06:
I will look into that, thanks.

RenanGarcia, 2018-03-08:
I'm still studying the library, but it seems that the problem was in fromRequest method ReadDiscreteInputsResponseBody and ReadCoilsResponseBody.
I solved the problem by creating functions that we can use in other parts of the lib. See if it's a good solution. #185

closed 2018-03-13 by RenanGarcia