Cloud-Automation/node-modbus #197: Documentation

robynjayqueerie, 2018-06-27:
Is there any documentation for jsmodbus apart from the readme and prowling through source? For instance I was looking for the events a server may emit and the format for returning data to the client.
jsmodbus looks very complete but a wiki or some documentation would come in handy
Thanks
Robyn

stefanpoeter, 2018-06-27:
Nope, the Readme is all there is. Feel free to add a list of events to the Readme, that would indeed be useful. What do you mean with "the format for returning data to the client"?

robynjayqueerie, 2018-06-27:
Thanks Stefan, as for the code to return data, just looking at the SimpleServer.js example you have the following
server.coils.writeUInt16BE(0x0000, 0)
server.coils.writeUInt16BE(0x0000, 2)
server.coils.writeUInt16BE(0x0000, 4)
server.coils.writeUInt16BE(0x0000, 6)

server.discrete.writeUInt16BE(0x5678, 0)

server.holding.writeUInt16BE(0x0000, 0)
server.holding.writeUInt16BE(0x0000, 2)

server.input.writeUInt16BE(0xff00, 0)
server.input.writeUInt16BE(0xff00, 2)
I expect those calls handlle the replies to the client. I think I have found that the hex value is a start address but otherwise have not been able to decode them.
Regards
Robyn

stefanpoeter, 2018-06-27:
Hi Robyn,

modbus is generally speaking a protocol to communicate via a shared memory. The Holding, Input, Coils and Discrete Buffers are different memory areas (implemented with node.js buffer). So the requests for those buffers (read-holding-registers, function code 3) for example reads from the holding buffers. If you apply values to the buffer like so

````
server.holding.writeUInt16BE(0x0001, 0)
server.holding.writeUInt16BE(0x0002, 2)
````
you write values to the holding registers so that clients can read the values 0x0001 and 0x0002 from the modbus registers 0 and 1 (since modbus adresses in 16bit length and the node.js buffer addresses in 8bit length)

If you want to react to certain events like a read-holding-register event then you can implement a handler for this event and work with the request directly.

robynjayqueerie, 2018-06-27:
Thanks for that, I had it the wrong way round, I thought the first value was the address and the second the data, now I think I can make better sense of it.
It is a good package a bit more documentation could make it shine.
Thanks heaps

closed 2018-06-27 by stefanpoeter