Cloud-Automation/node-modbus #199: 3 test failing (locally)?

martijnthe, 2018-06-30:
Hi Stefan,

I just checked out your repo and to try it out.
First thing I tried is running the unit tests.
Somehow 3 of them are failing (but it looks like they are not failing in CI!?). See below for the logs.
I'll dig into it to try to figure out why.

```
  Modbus/RTU Client Response Tests
    1) should handle a valid read coils response
    2) should handle a exception
    3) should handle a chopped response

...

  3 failing

  1) Modbus/RTU Client Response Tests should handle a valid read coils response:
     TypeError: Cannot read property 'address' of undefined
      at Context.<anonymous> (test/rtu-client-response-handler.test.js:31:30)

  2) Modbus/RTU Client Response Tests should handle a exception:

      AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:

  assert.ok(response !== undefined)

      + expected - actual

      -false
      +true

      at Context.<anonymous> (test/rtu-client-response-handler.test.js:51:12)

  3) Modbus/RTU Client Response Tests should handle a chopped response:

      AssertionError [ERR_ASSERTION]: The expression evaluated to a falsy value:

  assert.ok(response !== undefined)

      + expected - actual

      -false
      +true

      at Context.<anonymous> (test/rtu-client-response-handler.test.js:79:12)
```

martijnthe, 2018-06-30:
The first test is failing because of this line:
https://github.com/Cloud-Automation/node-modbus/blob/c27c8f10270cba5d134f80d8f67ba801c01b99c7/src/rtu-response.js#L22

The `bodyCount` property does not exist, so `1 + body.bodyCount` end up being `NaN`.

I searched for `bodyCount` in the repo, but this line is the only instance of the word I could find.

Any ideas?

stefanpoeter, 2018-06-30:
What version have you checked out? v3.0.0 is fine, there is a v3.0.1 that contains some untested code for modbus rtu.

martijnthe, 2018-06-30:
@stefanpoeter I tried `master` as well as `v3.0.1_dev`.
I've created a fix, PR coming in a minute.

stefanpoeter, 2018-06-30:
My tests are all fine. Tell me your node.js and npm version as well as your Operating System.

stefanpoeter, 2018-06-30:
````
  Buffer and status conversions.
    ✓ should convert 10 coils status to buffer 2 bytes
    ✓ should convert a buffer with hex to coils/discrete array status
    ✓ should convert a buffer with bin to coils/discrete array status
    ✓ should return empty array if buffer is not instance of Buffer
    ✓ should return empty buffer if coils is not instance of Array

  Buffer manipulation tests
    ✓ should return a short buffer
    ✓ should return a longer buffer
    ✓ should return a correct only byte
    ✓ should return a correct buffer
    ✓ should return another correct buffer

  ReadCoils Tests.
    ReadCoils Response
      ✓ should create a buffer from a read coils message
      ✓ should create a message object from a buffer
      ✓ should mask out extra bits
      ✓ should return an individual coil if requested
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code
      ✓ should return <55> when addres = 0 and count = 8 for coils <55 55 55>
      ✓ should return <55> when addres = 6 and count = 8 for coils <55 55 55>
      ✓ should return <55 01> when addres = 0 and count = 9 for coils <55 55 55>
      ✓ should return <2A> when addres = 1 and count = 7 for coils <55 55 55>
      ✓ should return <05> when addres = 0 and count = 4 for coils <55 55 55>
      ✓ should return <02> when addres = 1 and count = 3 for coils <55 55 55>
    ReadCoils Requests
      ✓ should create a buffer from a read coils message
      ✓ should create a message object from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  ReadDiscreteInputs Tests.
    ReadDiscreteInputs Response
      ✓ should create a buffer from a read discrete inputs message
      ✓ should create a message object from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code
    ReadDiscreteInputs Requests
      ✓ should create a buffer from a discrete inputs message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  ReadHoldingRegisters Tests.
    ReadHoldingRegisters Request
      ✓ should create a buffer from a read holding registers message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  ReadInputRegisters Tests.
    ReadInputRegisters Request
      ✓ should create a buffer from a read input registers message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  Modbus Response Tests.
    Read Coils Tests.
      ✓ should create request from buffer
      ✓ should handle invalid buffer content

  Modbus/RTU Client Request Tests
    Register Test.
      ✓ should register an rtu request
    Handle Data Tests.
[... 98 lines cut ...]
      ✓ should return null on wrong function code

  WriteMultipleRegisters Tests.
    WriteMultipleRegisters Request
      ✓ should create a buffer from a write multiple registers message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  WriteSingleCoil Tests.
    WriteSingleCoil Response
      ✓ should create a buffer from a write single coil message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  WriteSingleRegister Tests.
    WriteSingleRegister Request
      ✓ should create a buffer from a write single register message
      ✓ should create a message from a buffer
      ✓ should return null on not enough buffer data
      ✓ should return null on wrong function code

  128 passing (161ms)
````

stefanpoeter, 2018-06-30:
you are using node 10.3.0. Can you execute the tests with node 8?

martijnthe, 2018-06-30:
Hah, it works with Node 8.
But I still think there's a bug. See my PR https://github.com/Cloud-Automation/node-modbus/pull/200.
I suspect that the crc calculator is returning 0 when you give it `NaN`. And 0 happens to be your test's expectation for the crc value. I will take a look what's going on now.

martijnthe, 2018-06-30:
OK, mystery solved.
so `buffer.readUInt16BE(NaN)` returns `0` in Node 8, but in Node 10 it will throw.

Node 8
```
Buffer.prototype.readUInt16BE = function(offset, noAssert) {
  offset = offset >>> 0;  /// <---- NaN >>> 0 === 0
  if (!noAssert)
    checkOffset(offset, 2, this.length);
  return (this[offset] << 8) | this[offset + 1];
};
```

Node 10
```
function readUInt16BE(offset = 0) {
  checkNumberType(offset);
  const first = this[offset];
  const last = this[offset + 1];
  if (first === undefined || last === undefined)
    boundsError(offset, this.length - 2);

  return first * 2 ** 8 + last;
}
```

stefanpoeter, 2018-06-30:
Good work! Will be fixed in 3.0.1.

commit 688b2e83 (2018-06-30): Merge pull request #200 from martijnthe/fix-#199

Fixes #199 – RTU CRC failed because of typo bodyCount => byteCount
commit d0a8949e (2018-06-30): Fixes #199 – RTU CRC failed because of typo bodyCount => byteCount

The expectated CRC value in the test case itself also did not match the actual value.
I also changed the value to be able to detect byte swaps.
closed 2018-06-30 by stefanpoeter