Cloud-Automation/node-modbus #201: RTU Server?

martijnthe, 2018-06-30:
hi Stefan,

Looking at the code, RTU server is not supported today, correct?

Do you have some thoughts on how to restructure the code to add support for it?

I started reading through the code a bit. My notes so far:

- Factor out the many cases in https://github.com/Cloud-Automation/node-modbus/blob/master/src/tcp-server-response-handler.js#L21 and move them into a transport agnostic  `fromRequest(request, server)` method in `ResponseFactory` (https://github.com/Cloud-Automation/node-modbus/blob/master/src/response/response-factory.js#L16)
- Add a `createPayload` method to `ModbusRTUResponse`.
- ...

stefanpoeter, 2018-06-30:
It shouldn't be that hard. Just create a modbus-rtu-server.js and handle everything like it is done in the modbus TCP server. You need an RTU request handler and also, like you said, a createPayload method in the RTUResponse.

If I remember correctly than there is no handshake in RTU so you don't need to remember new clients.

martijnthe, 2018-07-03:
PR: https://github.com/Cloud-Automation/node-modbus/pull/204

closed 2018-07-11 by stefanpoeter