Cloud-Automation/node-modbus #203: On Register Change Event

wongjasont, 2018-07-02:
I am implementing a `jsmodbus` simulator. One of the key features I'm implementing is that one particular register will depend its value from another. In lieu:

I am looking for an event that the server could listen to when a particular register has been updated;
For example, `client.writeMultipleRegisters(0, [1,2,3,4,5])` has been invoked; a particular event I'm expecting should be triggered, and it should be emitting this event with the address like:

```
server.onRegisterChange( (address, changedValues) => {
     // LOGIC HERE
})
```

stefanpoeter, 2018-07-03:
You can listen to the pre/post events that the server emits for the single requests (unfortunatly a few events for certain fcs are missing) an check the memory areas related to the request and call your handler.

stefanpoeter, 2018-07-11:
Added pre and post events to all available function codes. See branch v3.1.0-dev

closed 2018-07-11 by stefanpoeter