Cloud-Automation/node-modbus #205: Unused/old code?

martijnthe, 2018-07-03:
This file doesn't appear to be used?:
- `src/modbus-server-core.js`

These examples look dated, they reference `ModbusClient.createSerialClient`, which doesn't seem to exist:
- `examples/serial/ReadDiscreteInput.js`
- `examples/serial/WriteSingleCoil.js`
- `examples/serial/ReadInputRegister.js`
- `examples/serial/WriteSingleRegister.js`

- `examples/tcp/AutoReconnect.js` -> references `modbus.client.tcp` which also doesn't seem to exist.

Suggestion:
- Make the examples part of an (integration) test that gets run as part of CI

stefanpoeter, 2018-07-03:
I like your suggestion with the examples as part of the tests and yes, those files are from v2. Should be removed with the next patch.

stefanpoeter, 2018-07-11:
Fixed the examples in branch v3.1.0-dev. Thanks for the hint.

closed 2018-07-11 by stefanpoeter