Cloud-Automation/node-modbus #210: Add timing based framing for RTU client and server

martijnthe, 2018-07-10:
The spec says:

> In RTU mode, message frames are separated by a silent interval of at least 3.5 character times.

(see [Modbus_over_serial_line_V1_02.pdf])

So, if data comes in, then a pause of longer than 3.5 character times and then more data, they should be considered to be separate frames. Right now this is not the case. I noticed this because I've got some noise on the lines and get stray bytes here and there. They end up getting reassembled with whatever comes next (of which parsing fails miserably...). I think the easiest way to implement this is to note the last time (incomplete) data came in. If new data comes, look at the last time. If it's over 3.5 character times, just ignore the data in the reassembly buffer.

[Modbus_over_serial_line_V1_02.pdf]: http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

stefanpoeter, 2018-07-11:
I don't think setting a timestamp is a good Idea. I think we should improve the plausibility tests. With TCP we would simply close a client connection when there is no plausible data, here I would just throw away the first byte when the current buffer head didn't make sense and start again.

martijnthe, 2018-07-11:
> I would just throw away the first byte when the current buffer head didn't make sense and start again.

Wouldn't that preclude the ability to receive payloads in multiple chunks?

stefanpoeter, 2018-07-11:
Have a look at tcp-server-request-handler.js, it is already handled there.

closed 2019-01-02 by stefanpoeter