Cloud-Automation/node-modbus #211: not enough data available

nathanbussey, 2018-07-10:
when I call client.WriteSingleRegister(), it fails the second time that tcp-response.js calls fromBuffer. It works fine the first time, but the second time it generates a RangeError in checkOffset of buffer.js.

I can't figure out why it's calling fromBuffer twice, but it succeeds fine the first time.

stefanpoeter, 2018-07-11:
Do you have some code for me to look at?

nathanbussey, 2018-07-11:
Your test code exercises this issue. And I don't think it's specifically an issue, but most of the calls seem to be returning null because they are calling that fromBuffer twice. I added in console logs in tcp-response.js so I could see when it was happening. and it happens a lot in the test, but I think it's as intended.

I install v3.1.0-dev and I was able to write single register and read single register.

nathanbussey, 2018-07-11:
I'm able to write 0x1234 to address 10 okay, but I'm having trouble writing to high addresses. Is WriteSingleRegister limited? I would have expected to be able to use between 0 and 0x270E.

stefanpoeter, 2018-07-11:
What server do you use? Modbus uses a full 16bit address space. If you use the SimpleServer Example then initiate the server with a bigger buffer.

nathanbussey, 2018-07-11:
I set the alloc to 20000 and it allowed me to use up to 270E as expected, if I'm treating that space correctly. 

Thanks for the help.

closed 2018-07-11 by stefanpoeter