Cloud-Automation/node-modbus #216: responseDelay for server

biancode, 2018-08-10:
Hi Stefan,

the response delay of the server is missing in my case in v3. That was a great option to simulate slow devices with the server. Is it possible to get that feature back into v3?

cya
Klaus

stefanpoeter, 2018-08-15:
Hi Klaus,
sorry for the late reply, i am pretty busy right now.

This feature should be easy to implement, feel free to make a pull request.

stefanpoeter, 2018-08-28:
I've gives this some thoughts: I think it is better to extend the socket module and overwride the send method to handle a response delay. This way you would have a common socket response module that is usable with jsmodbus. And this should not be that hard to implement.

closed 2018-08-28 by biancode