Cloud-Automation/node-modbus #220: Modbus server response handler has a minor bug

Alex-OPTIM, 2018-10-21:
src/modbus-server-response-handler.js at line [241](https://github.com/Cloud-Automation/node-modbus/blob/master/src/modbus-server-response-handler.js#L241) has "IF" statement which is always false. `responseBody` doesn't have property `address`.

I guess class `WriteMultipleRegistersResponseBody` needs to be modified to get `address` property or in the "IF" statement it needs to be property `start`.

Or even it should be:
`if (responseBody.start * 2 + request.body.values.length > this._server.holding.length) { ...`

Is it correct?

stefanpoeter, 2018-10-23:
Hi @Alex-OPTIM 

This must actually be the address from the request body. So it must look something like this `requestBody.address * 2 + ... length ... > ... length`

so you are partially right. Thanks for the hint, I'll prepare a pull request.

stefanpoeter, 2019-01-02:
Thanks for your help.

closed 2019-01-02 by stefanpoeter