Cloud-Automation/node-modbus #224: Serial gets data, but also times out

mainstreetmark, 2018-12-20:
Hello - 
I'm following your [ReadHoldingRegisters](https://github.com/Cloud-Automation/node-modbus/blob/master/examples/serial/ReadHoldingRegisters.js) example (which uses readInputRegisters, fyi) and i cannot get it to work.  Here is my output:

> connecting to /dev/ttyUSB1 { baudRate: 9600, parity: 'none', stopBits: 1, dataBits: 8 }
socket on open
  rtu-client-request-handler registrating new request +0ms
catch { '0': { err: 'Timeout', message: 'Request timed out' } }
  rtu-response-handler receiving new data +0ms
  rtu-response-handler buffer <Buffer 01 03 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 ... > +0ms
  rtu-response address 1 buffer <Buffer 01 03 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 ... > +0ms
  rtu-response-handler crc 51210 +3ms
  rtu-response-handler reset buffer from 85 to 0 +0ms
  rtu-response-handler not enough data available to parse +1ms
  rtu-client-request-handler new response coming in +5s
  rtu-client-request-handler something is strange, received a respone without a request +0ms
data { '0': <Buffer 01 03 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 20 00 00 c2 20 00 ... > }
close { '0': null }

As you can see, the buffer appears in the response, but the instant it does, it thinks it's timed out.
Incidentally, the device i'm reading from can successfully be read via [ModScan32](https://www.win-tech.com/html/demos.htm) and another tool from https://www.modbusdriver.com/, so i know the device itself seems to work.

stefanpoeter, 2018-12-21:
Can you post your code. Seems strange indeed.

mainstreetmark, 2018-12-21:
Sure.

[test2.js.txt](https://github.com/Cloud-Automation/node-modbus/files/2703300/test2.js.txt)

stefanpoeter, 2019-01-02:
You catch the socket data twice in your js file. Remove the line

```
socket.on('data', function () {
  console.log("data", arguments)
})
```

And try again.

mainstreetmark, 2019-01-02:
No difference.  :(

It prints "socket on open", waits 5 seconds and prints the rest of that stuff.

I can see in the debug buffer the value 0xC220, which is real data.  In fact, here's the same modbus request performed by [modpoll](https://www.modbusdriver.com/modpoll.html) (in 110ms, i should add):

> $ modpoll /dev/ttyUSB0 -b 9600 -p none -r 200 -c 40 -t 4:hex
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.
>
>Protocol configuration: Modbus RTU
Slave configuration...: address = 1, start reference = 200, count = 40
Communication.........: /dev/ttyUSB0, 9600, 8, 1, none, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register (hex), output (holding) register table
>
>-- Polling slave... (Ctrl-C to stop)
[200]: 0x0000
[201]: 0x0000
[202]: 0x0000
[203]: 0x0000
[204]: 0x0000
[205]: 0x0000
[206]: 0x0000
[207]: 0x0000
[208]: 0x0000
[209]: 0x0000
[210]: 0x0000
[211]: 0xC220
[212]: 0x0000
[213]: 0xC220
[214]: 0x0000
[215]: 0x0000
[216]: 0x0000
[217]: 0x0000
[218]: 0x0000
[219]: 0x0000
[220]: 0x0000
[221]: 0xC220
[222]: 0x0000
[223]: 0xC220
[224]: 0x0000
[225]: 0x0000
[226]: 0x0000
[227]: 0x0000
[228]: 0x0000
[229]: 0x0000
[230]: 0x0000
[231]: 0x0000
[232]: 0x0000
[233]: 0x0000
[234]: 0x0000
[235]: 0x0000
[236]: 0x0000
[237]: 0x0000
[238]: 0x0000
[239]: 0x0000

stefanpoeter, 2019-01-02:
Can you log a timestamp of the request and one for the catch and post the output. I want to see how much time has passed. The data seems to come in ok, just too late.

mainstreetmark, 2019-01-04:
>1546614143854 0 'connecting to' '/dev/ttyUSB0' { baudRate: 9600, parity: 'none', stopBits: 1, dataBits: 8 }
1546614143888 34 'socket on open'
1546614148908 5054 'catch' { '0': { err: 'Timeout', message: 'Request timed out' } }
1546614148916 5062 'close' { '0': null }

First number is Date.now(), second number is elapsed time since the "connecting to" event.

stefanpoeter, 2019-01-04:
Seems ok by me. The client does not receive data for five seconds and therefor returns an error. What operating system are you using? Maybe have a look at the serialport module, seems like the error is located there.

mainstreetmark, 2019-01-04:
Great.  Hopefully the #227 guy will report back on what sort of setup he's using.  Otherwise I'm stuck.

mainstreetmark, 2019-01-04:
Oh, Ubuntu 18

sivakatru, 2019-01-11:
Same issue at my side with Ubuntu 16.

closed 2019-09-17 by stefanpoeter