Cloud-Automation/node-modbus #227: Error after receiving the data

Vishal000, 2019-01-04:
Hi, 

I am receiving the intended output from the modbus library but after each data received I am getting the following output:
data <Buffer 30 31 30 33 30 39 30 31 30 33 30 34 37 46 46 46 46 46 46 46 64 32 36 37 34 63 35 61>
error from serial in { '0': { err: 'Timeout', message: 'Request timed out' } }

I am concerned with the error timeout. 

This is the following code that I have written:
let modbus = require('jsmodbus')
let SerialPort = require('serialport')
var host = '/dev/ttyS2';
var options = {
  baudRate: 19200,
  parity: 'none',
  stopBits: 1,
  dataBits: 8
};
console.log("connecting to", host, options);
let socket = new SerialPort(host,options); 

let client = new modbus.client.RTU(socket, 1)

socket.on('close', function () {
  console.log("close", arguments)
})

socket.on('open', function () {
  console.log("socket on open");
  client.readHoldingRegisters(200, 40)
    .then(function (resp) {
      console.log("resp", resp)
      socket.close()
    }).catch(function () {
      console.error("catch", arguments)
      socket.close()
    })
})

socket.on('data', function () {
  console.log("data", arguments)
})

socket.on('error', console.error)

Is this a library issue or any fault in my code ?
Please resolve.

stefanpoeter, 2019-01-04:
Seems to be related to #224 

Can you start your application with env DEBUG=* and post the output.

stefanpoeter, 2019-03-20:
Like the debug message is saying 

>   modbus-client received data +334ms
  rtu-response-handler receiving new data +0ms
  rtu-response-handler buffer <Buffer 30 31 30 33 30 39 30 31 30 33 30 34 37 46 46 46 46 46 46 46 64 32 36 37 34 63 35 61> +17ms
  rtu-response address 48 buffer <Buffer 30 31 30 33 30 39 30 31 30 33 30 34 37 46 46 46 46 46 46 46 64 32 36 37 34 63 35 61> +0ms
  response-factory fc 49 payload <Buffer 31 30 33 30 39 30 31 30 33 30 34 37 46 46 46 46 46 46 46 64 32 36 37 34 63 35 61> +0ms
  rtu-response-handler not enough data available to parse +54ms

Data has been received but could not be interpreted correctly (not enough data available). I haven't had time to look into the byte data, feel free to do so. Maybe they are using  ASCII Mode or Modbus/TCP over RTU which is not supported in this module. Could be that your device manual can tell you something about it.

closed 2019-09-17 by stefanpoeter