Cloud-Automation/node-modbus #229: More consistent response bodies, please

mainstreetmark, 2019-01-09:
FC1: 
>  ReadCoilsResponseBody {
        _fc: 1,
        _coils: <Buffer 02 00 00>,
        _numberOfBytes: 3,
        _valuesAsBuffer: <Buffer 02 00 00>,
        _valuesAsArray: [Array]
 }

FC4
> ReadInputRegistersResponseBody {
        _fc: 4,
        _byteCount: 40,
        _values: [Array],
        _bufferLength: 42,
        _valuesAsArray: [Array] 
}

One has `_coils`, the other `_values`.  Why not both values?  (though, `_valuesAsArray` helps).
One has `_numberOfBytes`, one has `_byteCount`.
One has `_bufferLength`, the other doesn't at all.

Consistent responses would make it easer to work with.

stefanpoeter, 2019-01-10:
Something that bothers me is that I didn't pay much attention on the array/buffer mixup in the response and request bodies. But there is no easy way to fix that without breaking some APIs.

The inconsistent naming should actually representing the naming in the specification, though I musst admit that doesn't worked out very well.

I am always happy to accept pull requests if it really bothers you that much, but keep in mind that the old APIs musst stay. They can be marked as deprecated thou.

mainstreetmark, 2019-01-10:
I think you can leave the existing fields as is (for compatibility) but also decide on a consistent response.  For example, include `_byteCount` in both responses, even though one also has `_numberOfBytes`.  And in the docs, just stop referencing `_numberOfBytes`.

closed 2019-03-20 by stefanpoeter