Cloud-Automation/node-modbus #230: ModbusTCPServer

vilelam, 2019-01-22:
Hello,
I'm creating a Modbus TCP Server with the following code and from a client, I'm connecting to the server and calling the readHoldingRegisters method; however, it seems that ModbusTCPServer is not yet developed to reply to readHoldingRegisters. Is that correct? Maybe I'm doing something wrong here, but it seems that the Server is never emitting the 'readHoldingRegisters' event. 

```
'use strict'

let net = require('net')
let modbus = require('jsmodbus')
let netServer = new net.Server()
let holding = Buffer.alloc(10000)
let server = new modbus.server.TCP(netServer, {
  holding: holding
})

server.on('connection', function (client) {
  console.log('New Connection')
})

server.on('readCoils', function (request, response, send) {
  /* Implement your own */

  response.body.coils[0] = true
  response.body.coils[1] = false

  send(response)
})

server.on('readHoldingRegisters', function (request, response, send) {

  /* Implement your own */

})

server.on('preWriteSingleRegister', function (value, address) {
  console.log('Write Single Register')
  console.log('Original {register, value}: {', address, ',', server.holding.readUInt16BE(address), '}')
})

server.on('WriteSingleRegister', function (value, address) {
  console.log('New {register, value}: {', address, ',', server.holding.readUInt16BE(address), '}')
})

server.on('writeMultipleCoils', function (value) {
  console.log('Write multiple coils - Existing: ', value)
})

server.on('postWriteMultipleCoils', function (value) {
  console.log('Write multiple coils - Complete: ', value)
})

/* server.on('writeMultipleRegisters', function (value) {
  console.log('Write multiple registers - Existing: ', value)
}) */

server.on('postWriteMultipleRegisters', function (value) {
  console.log('Write multiple registers - Complete: ', holding.readUInt16BE(0))
})

server.on('connection', function (client) {

  /* work with the modbus tcp client */

})

server.coils.writeUInt16BE(0x0000, 0)
server.coils.writeUInt16BE(0x0000, 2)
server.coils.writeUInt16BE(0x0000, 4)
server.coils.writeUInt16BE(0x0000, 6)

server.discrete.writeUInt16BE(0x5678, 0)

server.holding.writeUInt16BE(0x0000, 0)
server.holding.writeUInt16BE(0x0000, 2)

server.input.writeUInt16BE(0xff00, 0)
server.input.writeUInt16BE(0xff00, 2)

console.log(process.argv[2])
netServer.listen(process.argv[2] || 8502)
```

stefanpoeter, 2019-01-28:
Hey @vilelam 

you've started the server with the holding Buffer

```

let holding = Buffer.alloc(10000)
let server = new modbus.server.TCP(netServer, {
  holding: holding
})

``` 

In that case jsmodbus fires a 'preReadHoldingRegisters' and a 'postReadHoldingRegisters' event to manipulate the holding buffers up front or do something after the request has been handled.

If you start the modbus server without a holding buffer the readHoldingRegisters event will be fired.

stefanpoeter, 2019-01-28:
Of course you can. But keep in mind to be compatible to the current API specification and of course write tests.

vilelam, 2019-01-28:
is there any guide I can follow? I will probably need to learn and gain some confidence first.
I will try to start with some simple contribution, probably to the README documentation file

stefanpoeter, 2019-01-28:
You can find a lot on the GitHub Help Site. Regarding the tests, just inspect the existing ones. Pull Requests do not hurt since I am the one who decides to merge them.

closed 2019-01-28 by stefanpoeter