Cloud-Automation/node-modbus #232: 'TypeError: "value" argument is out of bounds' error  when handling request response

paulmccarthy, 2019-02-22:
I'm trying to read data from a [NDMeter Rail 350](https://www.ndmeter.co.uk/cube-rail-350.html) power meter via Modbus RTU. I'm using a Raspberry Pi Compute Model based system, with the following:

Node JS Version: 8.15.0
NPM Version: 6.4.1
OS Version (uname -a): Linux 13001 4.4.45-v7+ #19 SMP Fri Feb 10 12:54:17 CET 2017 armv7l GNU/Linux
OS Version (/etc/os-release): 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

The command is sent via over modbus, and data is returned, but I get the following error:

```
UnhandledPromiseRejectionWarning: TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1185:11)
    at Buffer.writeUInt8 (buffer.js:1233:5)
    at /home/pi/modbus-test/jsmodbus/node_modules/jsmodbus/src/response/read-input-registers.js:90:15
    at Array.forEach (<anonymous>)
    at ReadInputRegistersResponseBody.createPayload (/home/pi/modbus-test/jsmodbus/node_modules/jsmodbus/src/response/read-input-registers.js:89:18)
    at ModbusRTUClientRequestHandler.handle (/home/pi/modbus-test/jsmodbus/node_modules/jsmodbus/src/rtu-client-request-handler.js:48:79)
    at ModbusRTUClient._onData (/home/pi/modbus-test/jsmodbus/node_modules/jsmodbus/src/modbus-client.js:51:30)
    at emitOne (events.js:116:13)
    at SerialPort.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
```

The issue is caused by the following code starting on line 84 in` src/response/read-input-registers.js`:
```
createPayload () {
    const payload = Buffer.alloc(this.byteCount)

    payload.writeUInt8(this._fc, 0)
    payload.writeUInt8(this.length, 1)
    this._values.forEach(function (value, i) {
      payload.writeUInt8(value, 2 + i)
    })

    return payload
  }
```
This power meter returns 16 bit values, so the `payload.writeUint8` function throws the error when it encounters a value > 255.

I've tried changing the code to:

```
createPayload () {
    const payload = Buffer.alloc(this.byteCount)

    payload.writeUInt16BE(this._fc, 0)
    payload.writeUInt16BE(this.length, 1)
    this._values.forEach(function (value, i) {
      payload.writeUInt16BE(value, 2 + i)
    })

    return payload
  }
```

which does return the expected values, but the then I get another error:

```
{ err: 'crcMismatch',
     message: 'the response payload does not match the crc' } }
```
[... 185 lines cut ...]
  serialport/bindings/unixWrite Finished writing 8 bytes +0ms
  serialport/stream binding.write write finished +8ms
  client-request-handler request fully flushed, ( error: undefined ) undefined +17ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +5ms
  serialport/bindings/poller Polling for "readable" +32ms
  serialport/bindings/poller received "readable" +462ms
  serialport/binding-abstract read +472ms
  serialport/bindings/unixRead Starting read +465ms
  serialport/bindings/unixRead Finished read 69 bytes +0ms
  serialport/stream binding.read finished +467ms
  modbus-client received data +494ms
  rtu-response-handler receiving new data +0ms
  rtu-response-handler buffer <Buffer 01 04 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 06 03 e8 00 00 00 ... > +0ms
  rtu-response address 1 buffer <Buffer 01 04 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 06 03 e8 00 00 00 ... > +0ms
  response-factory fc 4 payload <Buffer 04 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 06 03 e8 00 00 00 00 ... > +0ms
  rtu-response-handler crc 30089 +5ms
  rtu-response-handler reset buffer from 69 to 0 +0ms
  rtu-response-handler not enough data available to parse +1ms
  rtu-client-request-handler new response coming in +501ms
  rtu-client-request-handler create crc from response <Buffer 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 00 00 02 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... > +1ms
  rtu-client-request-handler CRC does not match 30089 !== 35447 +0ms
  serialport/stream _read reading +13ms
  serialport/binding-abstract read +16ms
  serialport/bindings/unixRead Starting read +16ms
```

stefanpoeter, 2019-02-23:
Can you post the code where you call the readInputRegisters method.

paulmccarthy, 2019-02-25:
Thanks for replying. My code is below:
```
const modbus = require('jsmodbus');
const SerialPort = require('serialport');

const socket = new SerialPort('/dev/ttyUSB4', {
    baudRate: 19200
});

const client = new modbus.client.RTU(socket, 1);

socket.on('open', function () {
    console.log('socket open');

    client.readInputRegisters(19456, 32)
        .then((resp) => {
            console.log(resp.response.body.valuesAsArray);
            socket.end();
            process.exit();
        }).catch(function () {
            console.error('readInputRegisters error', arguments);
            socket.end();
            process.exit(1);
        });
});

socket.on('socket error', function() {
    console.error('socket error', arguments);
    process.exit(1);
});
```

stefanpoeter, 2019-02-26:
Thanks for the code.

Changing the writeUint8 into a writeUint16 does not do any good and doesn't make sense. It seems like the payload buffer is not initiated with the right size. Can you add a log message for the byteCount like below and try again?

```
createPayload () {
    console.log('BYTE_COUNT:', this.byteCount)
    const payload = Buffer.alloc(this.byteCount)

    payload.writeUInt8(this._fc, 0)
    payload.writeUInt8(this.length, 1)
    this._values.forEach(function (value, i) {
      payload.writeUInt8(value, 2 + i)
    })

    return payload
  }
```

stefanpoeter, 2019-03-20:
Any progress?

dpham314, 2019-04-18:
I had a similar issue with reading a 16 bit ModbusRTU register, but with read-holding-registers. The values when createPayload ran were:

this._values = [2019, 4]
this._valuesAsBuffer = [7, 227, 0, 4]
this.byteCount = 6

Since 2019 > 255, it gives an error. It works when I change this._values to this._valuesAsBuffer in the createPayload function.
```
  createPayload () {
    if (this._values instanceof Buffer) {
      let payload = Buffer.alloc(2)
      payload.writeUInt8(this._fc, 0)
      payload.writeUInt8(this._byteCount, 1)
      payload = Buffer.concat([payload, this._values])
      return payload
    }

    if (this._values instanceof Array) {
      const payload = Buffer.alloc(this.byteCount)
      payload.writeUInt8(this._fc, 0)
      payload.writeUInt8(this._byteCount, 1)
      this._valuesAsBuffer.forEach(function (value, i) {
        payload.writeUInt8(value, 2 + i)
      })

      return payload
    }
  }
```

So the createPayload function in read-input-registers would be
```
  createPayload () {
    const payload = Buffer.alloc(this.byteCount)

    payload.writeUInt8(this._fc, 0)
    payload.writeUInt8(this.length, 1)
    this._valuesAsBuffer.forEach(function (value, i) {
      payload.writeUInt8(value, 2 + i)
    })

    return payload
  }
```

dvankuren, 2019-08-07:
I also ran into this in read-holding-registers when the register value exceeded max UInt8.  The suggestion provided by @dpham314 indeed resolved the problem.

stefanpoeter, 2019-09-17:
@dvankuren Should be resolved in the lates version. Can you check that? Version 3.1.4

closed 2019-12-23 by stefanpoeter