Cloud-Automation/node-modbus #233: ECONNRESET

MrWackers, 2019-02-27:
```
const modbus = require('jsmodbus')
        const net = require('net')
        const server = new net.Server()
        const mbserver = new modbus.server.TCP(server)
         
        server.listen(502)
```

The Code block above works, but on client disconnect i get:

```
Error: read ECONNRESET
    at _errnoException (util.js:992:11)
    at TCP.onread (net.js:618:25)
```
This is just a strait disconnect I dont know why it would be causing errors.

the code that I'm really working with is 
```
var netServer = new net.Server();

            netServer.on('error', (err) => {
                console.log('erro!');
                throw err;
            });

            this.server = new modbus.server.TCP(netServer, {
                holding: holding
            });

            this.server.on('connection', function (client) {
                console.log('New Connection')
            });

            this.server.on('error', (err) => {
                console.log('erro!');
                throw err;
            });

            netServer.listen(502);
```
and i just cant seem to make it stop. `netServer.on('error'` will not catch it either.

stefanpoeter, 2019-03-01:
Are you executing any requests after the disconnect on the client?

MrWackers, 2019-03-01:
It was when the CLient disconnected from the server. there could be a request but i dont think so.

I ended up muting it by doing the following.

```
var netServer = new net.Server(function(s){
                s.on('error', (err) => {
                });
            });
```

stefanpoeter, 2019-03-01:
What client are you using?

MrWackers, 2019-03-14:
Intellution IGS Driver

stefanpoeter, 2019-03-20:
Can you snif the network traffic with wireshark or tcpdump so we get a better look at what is happening?

closed 2019-06-30 by stefanpoeter