Cloud-Automation/node-modbus #236: Excess console logging

aapiskukko, 2019-04-23:
Branch: master (ba2a6a1)
Location: modbus-server-response-handler.js, row 251.
Issue: Excess console.log row spams the log file with unused lines. 
Fix: Remove line containing 'console.log(request.body)' to prevent writing this unused logging.

closed 2019-06-19 by stefanpoeter

alpintrekker, 2019-09-11:
this is a real problem, when you have 200+ requests per second.
at least, it would be nice to change it to debug.

jan-kiszka, 2020-03-04:
Seems like that console.log is a debug leftover from 554c2f5aae0f2204bf840f6dc7e7980e90f842a4. Would someone submit a PR to remove it?

stefanpoeter, 2020-03-04:
@alpintrekker I totally oversaw your comment there, sorry for that. I am always happy when people notify me about errors and stuff, in this case I am more than happy if you prepare a pull request and resolve this of course. Help is always welcome.

alexbuczynsky, 2020-03-04:
Put in a PR to resolve this issue for v3. 

What is odd @jan-kiszka is that the `node-red-contrib-modbus` package as of 5.0.1 [has already upgraded to the 4.0 version](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/afe657b052c15e5addcc7b9b2a31896c311a0f55/package.json#L8) of the library and where the console.log statement has been removed already. So if you are using `node-red-contrib-modbus: 4.x.x` this problem should be fixed by the PR, but if you are using `node-red-contrib-modbus: 5.x.x` then this issue should already not be a problem.

jan-kiszka, 2020-03-04:
@alexbuczynsky Ah, good to know. We are on 4.1.3 with contrib-modbus, but we can ask the user to upgrade the node on the target (and we will do for the next release).