Cloud-Automation/node-modbus #238: Out of bounds

ortizvinicius, 2019-06-19:
Hi, i been using this library for Modbus communication, and when used via ethernet it works very well, but when i try to communicate to a CLP via Modbus RTU i got this error:
´´´
(node:20088) UnhandledPromiseRejectionWarning: TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1185:11)
    at Buffer.writeUInt8 (buffer.js:1233:5)
    at C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\ldl\commtest\node_modules\jsmodbus\src\response\read-holding-registers.js:99:17
    at Array.forEach (<anonymous>)
    at ReadHoldingRegistersResponseBody.createPayload (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\ldl\commtest\node_modules\jsmodbus\src\response\read-holding-registers.js:98:20)
    at ModbusRTUClientRequestHandler.handle (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\ldl\commtest\node_modules\jsmodbus\src\rtu-client-request-handler.js:48:77)
    at ModbusRTUClient._onData (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\ldl\commtest\node_modules\jsmodbus\src\modbus-client.js:51:30)
    at emitOne (events.js:121:20)
    at SerialPort.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
(node:20088) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or
by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:20088) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
´´´

This is the test code:
´´´
const modbus = require('jsmodbus')
const SerialPort = require('serialport')
const socket = new SerialPort('COM6', {
  baudRate: 4800
})

const client = new modbus.client.RTU(socket, 1)

socket.on('close', function () {
  console.log(arguments)
})

socket.on('open', function () {
  client.readHoldingRegisters(50020, 1)
    .then(function (resp) {
      console.log(resp.response)
      socket.close()
    }).catch(function () {
      console.error(arguments)
      socket.close()
    })
})

socket.on('data', function () {
  console.log(arguments)
})

socket.on('error', console.error)
´´´

stefanpoeter, 2019-06-30:
Hi @ortizvinicius,

since I don't own a PLC with a Serial Modbus Interface I cannot reproduce this error. Run your code with the DEBUG Environment Variable set to * to get some more informations.

```
DEBUG=* node yourapp.js
```

ortizvinicius, 2019-07-04:
There it is:

```
serialport/bindings loading WindowsBinding +0ms
  serialport/stream opening path: COM6 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/stream opened path: COM6 +6ms
  modbus-client issuing new read holding registers request +0ms
  rtu-client-request-handler registrating new request +0ms
  user-request creating new user request with timeout 5000 +0ms
  client-request-handler flushing +0ms
  client-request-handler flushing new request <Buffer 01 03 c3 64 00 01 f9 91> +1ms
  serialport/stream _write 8 bytes of data +5ms
  serialport/binding-abstract write 8 bytes +12ms
  serialport/stream _read reading +1ms
  serialport/binding-abstract read +1ms
  serialport/stream binding.write write finished +1ms
  client-request-handler request fully flushed, ( error: undefined ) undefined +5ms
  client-request-handler flushing +5s
  client-request-handler no request to be executed +1ms
{ '0': { err: 'Timeout', message: 'Request timed out' } }
  serialport/stream #close +5s
  serialport/binding-abstract close +5s
  serialport/stream binding.read error { Error: Reading from COM port (ReadIOCompletion): Operation aborted canceled: true } +4ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/stream binding.close finished +1ms
{ '0': null }

C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\LDL\CommTest>node RTUtest
  serialport/bindings loading WindowsBinding +0ms
  serialport/stream opening path: COM6 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +6ms
  serialport/stream opened path: COM6 +70ms
  modbus-client issuing new read holding registers request +0ms
  rtu-client-request-handler registrating new request +0ms
  user-request creating new user request with timeout 5000 +0ms
  client-request-handler flushing +0ms
  client-request-handler flushing new request <Buffer 01 03 c3 64 00 0a b8 56> +8ms
  serialport/stream _write 8 bytes of data +21ms
  serialport/binding-abstract write 8 bytes +96ms
  serialport/stream _read reading +3ms
  serialport/binding-abstract read +3ms
  serialport/stream binding.write write finished +3ms
  client-request-handler request fully flushed, ( error: undefined ) undefined +12ms
  serialport/stream binding.read finished +31ms
  modbus-client received data +61ms
  rtu-response-handler receiving new data +0ms
  rtu-response-handler buffer <Buffer 01 03 14 00 00 01 0e 00 00 00 00 7f ff 00 00 00 00 7f ff 00 00 00 00 ef 5f> +0ms
  rtu-response address 1 buffer <Buffer 01 03 14 00 00 01 0e 00 00 00 00 7f ff 00 00 00 00 7f ff 00 00 00 00
ef 5f> +0ms
  response-factory fc 3 payload <Buffer 03 14 00 00 01 0e 00 00 00 00 7f ff 00 00 00 00 7f ff 00 00 00 00 ef
5f> +0ms
  ReadHoldingRegistersResponseBody ReadHoldingRegistersResponseBody values [ 0, 270, 0, 0, 32767, 0, 0, 32767, 0, 0 ] +0ms
  rtu-response-handler crc 24559 +7ms
  rtu-response-handler reset buffer from 25 to 0 +1ms
  rtu-response-handler not enough data available to parse +1ms
  rtu-client-request-handler new response coming in +70ms
(node:13412) UnhandledPromiseRejectionWarning: TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1185:11)
    at Buffer.writeUInt8 (buffer.js:1233:5)
    at C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\LDL\CommTest\node_modules\jsmodbus\src\response\read-holding-registers.js:99:17
    at Array.forEach (<anonymous>)
    at ReadHoldingRegistersResponseBody.createPayload (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\LDL\CommTest\node_modules\jsmodbus\src\response\read-holding-registers.js:98:20)
    at ModbusRTUClientRequestHandler.handle (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\LDL\CommTest\node_modules\jsmodbus\src\rtu-client-request-handler.js:48:77)
    at ModbusRTUClient._onData (C:\Users\vinic\Documents\Termoplex\Projetos\RMSC\src\LDL\CommTest\node_modules\jsmodbus\src\modbus-client.js:51:30)
    at emitOne (events.js:121:20)
    at SerialPort.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
(node:13412) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:13412) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
  client-request-handler flushing +5s
  client-request-handler no request to be executed +1ms
{ '0': { err: 'Timeout', message: 'Request timed out' } }
  serialport/stream #close +5s
  serialport/binding-abstract close +5s
  serialport/stream binding.close finished +2ms
{ '0': null }
```

New test code:

```
const modbus = require('jsmodbus')
const SerialPort = require('serialport')
const socket = new SerialPort('COM6', {
  baudRate: 19200,
  Parity: 'none',
  stopBits: 1,
  dataBits: 8
})

const client = new modbus.client.RTU(socket, 1)

socket.on('close', function () {
  console.log(arguments)
})

socket.on('open', function () {
  client.readHoldingRegisters(50020, 10)
    .then(function (resp) {
      console.log(resp.response)
      socket.close()
    }).catch(function () {
      console.error(arguments)
      socket.close()
    })
})

socket.on('data', function () {
  console.log(arguments)
})

socket.on('error', console.error)
```

stefanpoeter, 2019-09-17:
@ortizvinicius Can you check if this is still an issue with v3.1.4? Thanks!

ortizvinicius, 2019-09-17:
Hi @stefanpoeter 
I don´t have the PLC anymore :S 
I will try to get one and do the test

stefanpoeter, 2019-10-16:
@ortizvinicius any progress?

closed 2019-12-23 by stefanpoeter