Cloud-Automation/node-modbus #242: v4.0 Checklist

alexbuczynsky, 2019-07-10:
Wanted to consolidate v4.0 discussions and create a common todo list

## Todos:
- [ ] Provide Typescript examples to compliment javascript examples
- [ ] Update Javascript examples if needed
- [ ] Make sure that the `valuesAsArray` and `valuesAsBuffer` never return undefined (affects requests and responses)

alexbuczynsky, 2019-07-10:
- [ ] test

alexbuczynsky, 2019-07-10:
@stefanpoeter I know you have a couple of items you wanted to add to the 4.0 milestone. Any thoughts on those?

stefanpoeter, 2019-07-10:
A quick one before Work catches me:

- Command Line Requests like jsmodus fc03 0:10 --repeat=50 (ms). for example

I'll come Back to this

stefanpoeter, 2019-07-15:
I added some items to the list. Unfortunatly work gets the better of me right now so I'll won't have much time.

alexbuczynsky, 2019-07-18:
I have been caught up with work as well.

Which function codes did you want to tackle?

alexbuczynsky, 2019-08-18:
@stefanpoeter I completed a couple of the items I wanted to get to such as creating working examples in typescript and making sure that the `_valuesAsBuffer` and `valuesAsArray` types are always set correctly. 

For the CLI tool, were you thinking of using something like [commander](https://www.npmjs.com/package/commander)?

As far as additional function codes, I would assume you are talking about FC 7-8, 11-12, 17, 20-24, etc. as listed here [Modbus Application Protocol Spec V1.1b](http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf)? If so do you want this for 4.0 or for future version such as 4.1? (on a selfish note, I would like to get 4.0 out so that my department can start using the typescript based version in production)

stefanpoeter, 2019-08-28:
Hi @alexbuczynsky 

sorry for the late reply.

I would use command for the CLI tools thats right.

Regarding the other FCs, if you are in a hurry these can be added in later minor versions. Unfortunatly I am still busy with other projects and there is currently not that much time left for this version on my side.

alexbuczynsky, 2019-10-16:
@stefanpoeter 

I added the changes you made from #244 into the 4.0 branch. Just fyi, it won't be as clean of a merge when we bring the 4.0 branch into the master, since the new code base is in typescript.

I added a basic cli based on what you were describing. It only does read requests over TCP. The template is there, so if you want to add requests for RTU or write requests feel free. Here are the options when you type `jsmodbus --help`

```sh
Usage: jsmodbus [options] [command]

Options:
  -V, --version                           output the version number
  -h, --help                              output usage information

Commands:
  fc01 [options] <host> <unitId> <range>  Read Coils
  fc02 [options] <host> <unitId> <range>  Read Discrete Inputs
  fc03 [options] <host> <unitId> <range>  Read Holding Registers
  fc04 [options] <host> <unitId> <range>  Read Input Registers
```

Here are the options for the function calls (`jsmodbus fc03 --help`)
```sh
Usage: jsmodbus fc03 [options] <host> <unitId> <range>

Read Holding Registers

Options:
  -p, --port <port>        Modbus Port (502 by default)
  -r, --repeat <time>      Repeat interval in milliseconds
  -b, --buffer             Print output as buffer
  -t, --timeout <timeout>  connection timeout in milliseconds (2000 by default)
  -bm, --benchmark         show timestamp benchmark information
  -h, --help               output usage information
```

stefanpoeter, 2019-10-16:
Hey @alexbuczynsky,

since you've been investing some time in this i would say it is more than fair to publish this as 4.0. Since I need some time to get used to typescript I would ask you to take care of issues concerning this version. Is that ok for you?

stefanpoeter, 2019-10-16:
And by the way, the CLI looks awesome. Great work. :-)

stefanpoeter, 2019-10-16:
I am not sure we need to bring this branch into the master. I would simply tag this as v4.0.0 and I assume npm does the rest. So we do not need to merge this into the master branch. But I'll check that.

stefanpoeter, 2019-10-16:
I've just published a v3.1 version from the v3.1-dev branch. It seems to work so the v4.0-dev branch is the place to put your changes in and I'll tag and publish it then.

stefanpoeter, 2019-10-16:
What about the Standard Code Style in the README. Is there anything like it for TypeScript?

alexbuczynsky, 2019-10-16:
However you want to handle the release cycle / branch management is for sure your prerogative. Doesn't matter to me as long as NPM can handle the changes :)

Yes, I would be fine helping with issues concerning the 4.0 release. I will probably defer to you on when I am not sure about the underlying original code base in cases where I am not sure how it was originally programmed.

Standardjs actually supports typescript allegedly... See here https://standardjs.com/#can-i-use-a-javascript-language-variant-like-flow-or-typescript. We would have to use eslint as the linter and install the plugin as they suggest on their website. I would have to test it out first and then get back to you.

alexbuczynsky, 2019-10-16:
I figured out how to add the linting standards. I am using two packages `tslint` and `tslint-config-standard` to apply the linting. The linting is applied before running the test / watch scripts. I then had to fix all the linting errors in the typescript code which took a hot minute. Anyway we can keep the badge on the readme.

alexbuczynsky, 2019-10-16:
I will run a couple checks to make sure that the npm package will compile both the commander file and the dist code base correctly. I will let you know once I have checked the tarball file is correctly packaged and then I should be good to go on my end.

alexbuczynsky, 2019-10-16:
Had to add the bin folder to the final package. It should be good to go now!

stefanpoeter, 2019-10-16:
It's done :-)

closed 2019-12-23 by stefanpoeter