Cloud-Automation/node-modbus #247: Modbus RTU Continuously Time Out

shengxio, 2019-10-03:
Hello team:

I have been using this jsmodbus package to connect with a PLC unit from EXOR international. Originally, my coworker managed to use the TCP portion of the package which worked flawlessly with the PLC. However, when I was trying to switch the connection to the RTU side, it started to become my nightmare.

Here is my TCP portion of the code.

`const Modbus = require('jsmodbus');
const net = require('net');

function connectModbusTcp(host, port) {
  const modbusSocket = new net.Socket();
  const client = new Modbus.client.TCP(modbusSocket);
  const options = {
    host,
    port
  };

  modbusSocket.on('connect', () =>
    console.log('Connection through modbus established')
  );

  modbusSocket.on('error', console.error);

  modbusSocket.connect(options);

  return client;
}

module.exports = connectModbusTcp;
`

And here is my RTU portion of the code.

`const Modbus = require('jsmodbus');
const SerialPort = require('serialport');

function connectModbusRtu(baudRate,COMport, parity,dataBits,stopBits) {
  const options = {
    baudRate:baudRate,
    dataBits:dataBits,
    stopBits:stopBits,
    Parity:parity
  };
  const modbusPort = new SerialPort(COMport, options);
  const client = new Modbus.client.RTU(modbusPort, 1);

  modbusPort.on('open', function(){
	
	if(modbusPort.isOpen){
		console.log('Connection through modbus established.');
		/*
		client.readHoldingRegisters(1, 1)
		.then(function (err,resp) {
			console.log('I got a response.');
			console.log(resp);
			//modbusSocket.close();
		}).catch(function (err) {
			console.log('I got a error.');
			console.log(err);
			//console.error(arguments);
			//modbusSocket.close();
		});*/
		client.writeSingleRegister(1,12)
		.then(function(resp){
			console.log('I wrote the answer.');
			console.log(resp);
		}).catch(function(err){
			console.log('I had it wrong.');
			console.log(err);
		});
	}else{
		console.log('COMport open failure.');
	}
	
  })
  
    
  
  //modbusPort.write('come in buddy\n');
  
  modbusPort.on('data', function () {
	console.log('------------Data------------');
	console.log(arguments);
})
  
  modbusPort.on('close',function() {
	console.log('------------CLose------------');
	console.log(arguments);
	modbusPort.open();
  });

  modbusPort.on('error', function() {
	console.log('------------General Error------------');
	console.error;
  });
  
  //modbusPort.on('connect',console.log('Connection through modbus established'));
 
 
 
  return client;
}

module.exports = connectModbusRtu;
`

Both of them were called in the main code as following.

`const express = require('express');
const connectModbusTcp = require('../../modbus/connectModbusTcp');
const connectModbusRtu = require('../../modbus/connectModbusRtu');

const router = express.Router();

router.post('/', (req, res) => {
  console.log('Establishing modbus connection');
  const client = connectModbusRtu(9600,'COM4','NONE',8,1);
  //const client = connectModbusTcp('192.168.1.124', 502);
  req.app.set('modbusClient', client);
  res.json({ msg: 'Establishing modbus connection' });
});

module.exports = router;
`

However for the RTU portion, I would occasionally receive certain information from the PLC after the unit boot up, but the rest of the time just straight time out from both sides. 
Apparently I have checked with another Modbus simulation software MODBUS SIM32, and it was able to read and write to the PLC. Which only left me in even deeper confusion.

Did I miss anything? And is there any way to monitor the serial com port just to see what has been passed to each side?

Thanks!

Roland Ding

shengxio, 2019-10-03:
And sorry for the terrible code format. Just couldn't get the line properly returned.

stefanpoeter, 2019-10-03:
Well since your Code is far from readable I suggest you call your Script with the DEBUG Environment variable set to * to get some information on what is happening. If you Post the Output and format your Code I might have a closer look.

shengxio, 2019-10-03:
Ok, just fixed the code format to make it readable. With that being said, I haven't much experience with the DEBUG Environment variable. How to play with it? any recommended starting point?

shengxio, 2019-10-04:
Here is the output which I got from the debug environment:

{ err: 'Timeout', message: 'Request timed out' }
  express:router dispatching GET /api/registers?number=0&dataType=bool +4s
  express:router query  : /api/registers?number=0&dataType=bool +24ms
  express:router expressInit  : /api/registers?number=0&dataType=bool +1ms
  express:router jsonParser  : /api/registers?number=0&dataType=bool +2ms
  body-parser:json skip empty body +4s
  express:router urlencodedParser  : /api/registers?number=0&dataType=bool +2ms
  body-parser:urlencoded skip empty body +2ms
  express:router serveStatic  : /api/registers?number=0&dataType=bool +3ms
  send stat "C:\Users\cassiechenx\Google Drive\20 Business\Dynamic Innovations\Scully\RTU\client\dist\api\registers" +4s
  express:router <anonymous>  : /api/registers?number=0&dataType=bool +33ms
  express:router trim prefix (/api/registers) from url /api/registers?number=0&dataType=bool +12ms
  express:router router /api/registers : /api/registers?number=0&dataType=bool +2ms
  express:router dispatching GET /?number=0&dataType=bool +21ms
  modbus-client issuing new read holding registers request +4s
  rtu-client-request-handler registrating new request +4s
  user-request creating new user request with timeout 5000 +4s
  client-request-handler flushing +785ms
  client-request-handler executing another request, come back later +21ms
  client-request-handler flushing +4s
  client-request-handler flushing new request <Buffer 01 03 00 00 00 14 45 c5> +4ms
{ err: 'Timeout', message: 'Request timed out' }
  client-request-handler flushing +5s
  client-request-handler no request to be executed +24ms
{ err: 'Timeout', message: 'Request timed out' }

stefanpoeter, 2019-10-04:
Dude, really? Making your code bold doesn't make it better readable.

Like the debug messages are telling you:

> client-request-handler flushing new request <Buffer 01 03 00 00 00 14 45 c5> +4ms

This issue seems to be with the serialport module.

shengxio, 2019-10-05:
LOL, well, that is my best shot. Actually, I don't even know why it would end up as bold text LOL.
It is good to know that the jsmodbus was performing as it intended. cannot see any debug information from the serial port side though. I guess I probably should go and bug them then?

closed 2019-10-06 by stefanpoeter