Cloud-Automation/node-modbus #250: modbus : discover all the device Master and slave and generating a json file with all the devices and data model (parameters supporting)

yboujraf, 2019-12-23:
Dear @martijnthe , @stefanpoeter  , @marcosps , @michikuehne , @timelesshaze 

I am working on a proprietary protocol translator from /to modbus.

I need to discover all the modbus device master/slave and get the list of parameters they are able to support (finally the data model) for each devic discovered.

It's our data model for one parameter it could be string, boolean, real, integer, ....

{

                    "number": 0,
                    "identifier": "ParameTypeInteger",
                    "description": "",
                    "value": 0,
                    "minimum": -4096,
                    "maximum": 0,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 32,
                    "isOnline": true,
                    "formula": "1*\n1/",
                    "default": -1024,
                    "type": "integer"
                }               {

                    "number": 1,
                    "identifier": "ParameTypeReal",
                    "description": "",
                    "value": 0,
                    "minimum": 0,
                    "maximum": 4096,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 1,
                    "isOnline": true,
                    "formula": "10*\n10/",
                    "default": 100,
                    "type": "real"
                },
                {
                    "number": 2,
                    "identifier": "ParameTypeString",
                    "description": "",
                    "value": "text",
                    "access": "readWrite",
                    "enumeration": "Yes\nNo\nOk",
                    "isOnline": true,
                    "default": "default",
                    "type": "string"
                },
                {
                    "number": 3,
                    "identifier": "ParameTypeEnumeration",
                    "description": "",
                    "value": 0,
                    "access": "readWrite",
                    "enumeration": "Line\nMic+48V\nMic",
                    "isOnline": true,
                    "default": 0,
                    "type": "enum"
                },
                {
                    "number": 4,
                    "identifier": "ParameTypeBoolean",
                    "description": "",
                    "value": true,
                    "access": "readWrite",
                    "isOnline": true,
                    "default": false,
                    "type": "boolean"
                }
How we could do with you code to generate a json file with all the devices ?

Best regards
Youssef

stefanpoeter, 2019-12-23:
I absolutely have no idea what you want to do.

martijnthe, 2019-12-23:
I doubt that it is possible because the modbus protocol has no / very limited service discoverability features built in.

closed 2020-03-04 by stefanpoeter

marcosps, 2020-03-04:
@yboujraf sorry but I don't work with modbus anymore, that was just a freelance that I did some time ago =(