Cloud-Automation/node-modbus #251: static unitId

simsimo, 2019-12-30:
unitId is static i can't change it once it set. 
In the case we are communicating with multiple slave behind gateway we can't do it because unitId is static.

alexbuczynsky, 2019-12-31:
Hi @simsimo.

The reason the property is readonly, is that since each client has its own queue of requests, changing the unitId when the queue is not empty can result in unintended consequences due to the asynchronous nature of the requests.

The library is intended to use a multiple client approach. See the following as an example with communicating to a single host but with three slaves [1, 2, 5].

```ts
import Modbus from 'jsmodbus'
import { Socket, SocketConnectOpts } from 'net'

const socket = new Socket()

const options: SocketConnectOpts = {
  host: '127.0.0.1',
  port: 8502
}

const clientIds = [1, 2, 5];

const clients: Modbus.ModbusTCPClient[] = []

for (const unitId of clientIds) {
  clients.push(new Modbus.client.TCP(socket, unitId))
}

socket.on('connect', async () => {

  for (const client of clients) {
    try {
      const { metrics, request, response } = await client.readCoils(0, 5);
      console.log(`Client Id ${client.unitId}`)
      console.log('Transfer Time: ' + metrics.transferTime)
      console.log('Response Body Payload: ' + response.body.valuesAsArray)
      console.log('Response Body Payload As Buffer: ' + response.body.valuesAsBuffer)
    } catch (err) {
      console.error(err);
    }
  }

  // finally close the socket connection
  socket.end();
})

socket.connect(options)
```

simsimo, 2020-01-02:
i will go like that thank you for your help :)

closed 2020-01-02 by simsimo

alexbuczynsky, 2020-01-04:
@simsimo No problem. Glad I could help!