Cloud-Automation/node-modbus #252: rtu-client is offline if socket is open before construction

hansmaad, 2020-01-17:
If I pass an open `SerialPort` to `Modbus.client.RTU` all requests will fail, because the request handler missed the open event.

This happened to me the first time I tried to use jsmodbus. It's easy to workaround in normal cases but now I have the need to pass an open socket to multiple protocol handlers.

https://github.com/Cloud-Automation/node-modbus/blob/e57681a6e26990835533186d7baaeae143640f97/src/rtu-client-request-handler.js#L18-L25

Would it be possible to check `socket.isOpen` in the constructor to set ` this._state = 'online'` ?

```javascript
if (this._socket.isOpen) {
  this._onConnect();
} else {
  this._socket.on('open', this._onConnect.bind(this))
}
```

stefanpoeter, 2020-01-17:
Hi @hansmaad,

i'll guess you are using jsmodbus v3? Seems like a valid request. If you prepare a pull request with the latest v3 version I'll check and merge it as soon as possible.

hansmaad, 2020-01-17:
As it turned out... I'm currently on 4.0.0. But it doesn't seem to make a big difference regarding this issue...?
What is the status of v4? Is it not really an official release?

stefanpoeter, 2020-01-17:
v4 is an official release. You can patch it up there if you like.

alexbuczynsky, 2020-01-17:
I opened a pull request that I believe addresses the issue @hansmaad is observing. If this is the case, @stefanpoeter can you look over this and then bump the version to 4.0.1?

stefanpoeter, 2020-01-20:
If @hansmaad gives his ok I sure will :-)

hansmaad, 2020-01-20:
@stefanpoeter Looks good!

closed 2020-01-25 by alexbuczynsky