Cloud-Automation/node-modbus #256: Error messages should contain an request object (like in await readHoldingRegisters(..))

vogsphar, 2020-02-12:
On success (normal resp) i am not interested in the request, but when errors occur it would be good to know what was send to the Server.

alexbuczynsky, 2020-02-13:
Can you give an example to describe what you are talking about?

mknj, 2020-02-20:
It would be nice to have an e.request in the same way as a.request.
```
$ cat client.js 
const modbus = require('jsmodbus')
const net = require('net')
const socket = new net.Socket()
const options = { 'host': '127.0.0.1', 'port': '8502' }
const client = new modbus.client.TCP(socket)

socket.on('connect', function () {
  client.readInputRegisters(0, 2).then(a=> {
    console.log("request object inside successful return object",a.request)
    client.readInputRegisters(0, 3242).catch(e=>{
      console.log("very limited info in error object",e)
      socket.end()
    })
  })
})
socket.connect(options)
$ node client.js 
request object inside successful return object ModbusTCPRequest {
  _id: 1,
  _protocol: 0,
  _length: 6,
  _unitId: 1,
  _body:
   ReadInputRegistersRequestBody { _fc: 4, _start: 0, _count: 2 } }
very limited info in error object Error: InvalidQuantity
    at new ReadInputRegistersRequestBody (/home/bob/js/x/node_modules/jsmodbus/dist/request/read-input-registers.js:41:19)
    at ModbusTCPClient.readInputRegisters (/home/bob/js/x/node_modules/jsmodbus/dist/modbus-client.js:66:23)
    at client.readInputRegisters.then.a (/home/bob/js/x/client.js:10:12)
    at process._tickCallback (internal/process/next_tick.js:68:7)
```

vogsphar, 2020-02-21:
A successfull response contains `{metrics,request,response}`
The UserRequestError Object only contains `{err,message,response}`
It would be helpfull to also add request (and metrics?) to UserRequestError  `{err,message,request,response}`.

```
UserRequestError {
  err: 'ModbusException',
  message: 'A Modbus Exception Occurred - See Response Body',
  response: ModbusTCPResponse {
    _id: 2,
    _protocol: 0,
    _bodyLength: 3,
    _unitId: 24,
    _body: ExceptionResponseBody { _fc: 3, _code: 2 }
  }
}
```

```
{
  metrics: UserRequestMetrics {
    createdAt: 2020-02-21T07:56:33.867Z,
    startedAt: 2020-02-21T07:56:33.867Z,
    receivedAt: 2020-02-21T07:56:33.872Z
  },
  request: ModbusTCPRequest {
    _id: 1,
    _protocol: 0,
    _length: 6,
    _unitId: 24,
    _body: ReadHoldingRegistersRequestBody { _fc: 3, _start: 0, _count: 2 }
  },
  response: ModbusTCPResponse {
    _id: 1,
    _protocol: 0,
    _bodyLength: 7,
    _unitId: 24,
    _body: ReadHoldingRegistersResponseBody {
      _fc: 3,
      _byteCount: 4,
      _values: [Array],
      _bufferLength: 6,
      _valuesAsArray: [Array],
      _valuesAsBuffer: <Buffer 43 c8 01 48>
    }
  }
}
```

stefanpoeter, 2020-03-31:
> A successfull response contains {metrics,request,response}

Where is this happening?

alexbuczynsky, 2020-03-31:
@stefanpoeter that is in v4.0

Here is an example:
https://github.com/Cloud-Automation/node-modbus/blob/88bf1c20c75740b10e9adea8e6dcdc70795604a7/examples/typescript/tcp/ReadHoldingRegisters.ts#L16-L21

stefanpoeter, 2020-03-31:
Ah, I never noticed :-) Well @vogsphar is right, it totally makes sense that this is in the error response as well. I don't think it would corrupt the API. I'll put it on this weeks todo list.

alexbuczynsky, 2020-03-31:
Sounds good. Adding it should be fine. I can tackle adding it to the 4.0 version.

closed 2023-12-19 by stefanpoeter