Cloud-Automation/node-modbus #260: Where is API?

JianJD, 2020-03-31:
Is there an API document for this？

stefanpoeter, 2020-03-31:
Hi @JianJD ,

by now there is only the readme. A detailed API documentation is still on my bucket list.

closed 2020-03-31 by stefanpoeter

JianJD, 2020-03-31:
When can you finish this API document？I really need your package now ！！！Thank you

stefanpoeter, 2020-03-31:
The readme deals with the basics for this library and it should be no problem to go on from there.

However, if you want to participate in this free and open library then you are welcome to commit a more detailed documentation yourself.

JianJD, 2020-04-01:
Does this package have CRC verification?

stefanpoeter, 2020-04-01:
Modbus/RTU uses CRC verification and so does this package. Modbus/TCP does not since it is part of TCP.

jacobq, 2020-05-13:
The server examples don't appear to show how to handle any requests. Am I missing something / not looking in the right place? For example, how can I make a server that responds to a `readCoils` request from the client in the other example?

stefanpoeter, 2020-05-15:
Hey @jacobq 

of course I take a PR for this.