Cloud-Automation/node-modbus #262: Mistake in Readme.md

anxorl, 2020-04-23:
In section 'Server example' same variable is used as net.Server object and modbus.server.TCP object.

So, in the last line, when listen function is called, this code fail because it is an unknown function of modbus.server.TCP as it is a net.Server function.

Regards

jacobq, 2020-05-13:
Yeah, trying to run this via copy/paste results in something like this:

```
const server = new modbus.server.TCP(server)
      ^

SyntaxError: Identifier 'server' has already been declared
[90m    at wrapSafe (internal/modules/cjs/loader.js:1047:16)[39m
[90m    at Module._compile (internal/modules/cjs/loader.js:1097:27)[39m
[90m    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1153:10)[39m
[90m    at Module.load (internal/modules/cjs/loader.js:977:32)[39m
[90m    at Function.Module._load (internal/modules/cjs/loader.js:877:14)[39m
[90m    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)[39m
[90m    at internal/main/run_main_module.js:18:47[39m
```

stefanpoeter, 2020-05-15:
See the latest versions on branch v3.1 and branch v4.0.

closed 2020-05-15 by stefanpoeter