Cloud-Automation/node-modbus #271: [Server] How to check values before they are written to holding registers

Excodibur, 2020-09-08:
Hi,

is there a way to check values and potentially reject them before they are written to a holding register on server-side?

Taking a look at [the source code](https://github.com/Cloud-Automation/node-modbus/blob/f348a56c9e0a33149d7c465abfbb7abcdf7e9579/src/modbus-server-response-handler.ts#L335) I can see quite a few events are fired, but it is unclear to me how to interfer with the data being written to the register. Are the events (pre, post, etc) mostly there to write some status to the logs, or can they be used to affect how and if a request is processed?

Excodibur, 2020-09-08:
Nevermind. Maybe the fault is on my side. Modbus spec states: `the MODBUS protocol is unaware of the significance of any particular value of any particular register`. So I guess the server does not care by design what is written to the register, as long as it fits.
Being unfamiliar with Modbus the idea that a server does not care about what data can be assigned to its registers (representing different functions) is still weird to me.

stefanpoeter, 2020-09-09:
Well it is a basic communication by shared memory principle. That is a basic as it gets :-)

closed 2020-09-09 by stefanpoeter