Cloud-Automation/node-modbus #272: Reading Multiple RTU Slave Devices

komorebi-san, 2020-09-10:
Hi,

Is there a way to correctly read multiple slave devices in series, with different baudRate and parity?

For example:

ID: 10
Baurate: 9600 
Parity : None
Stopbits: 1

ID: 15
Baurate: 19200 
Parity : Even
Stopbits: 0

ID: 17
Baurate: 9600 
Parity : Even
Stopbits: 1

So the program will loop through each config and set according to the setting.

stefanpoeter, 2020-09-10:
I don't know if there is a 'correct' way but simple as it is, just create multiple clients with the named settings and read them. :-)

komorebi-san, 2020-09-10:
@stefanpoeter Currently, i am creating and initializing new clients for each settings in series. But it seems that opening and closing the ports is not ideal in terms of speed.

stefanpoeter, 2020-09-10:
That seems to me like a question for the guys from the serialport module.

closed 2021-05-19 by stefanpoeter