Cloud-Automation/node-modbus #280: Got an  error -  could not find a declaration file for module 'serialport'

vn7n24fzkq, 2021-02-20:
Hi, thanks for  this awesome lib.
I got an error - " could not find a declaration file for module 'serialport' " when I build a typescript project.

I am using `jsmodbus:4.0.5` and I also check the package-lock.json.
The require block on contains `crc : 3.4.0` and `debug : ^3.1.0`, I think `serialport` should put in dependencies too, not just in  devDependencies.🤔

alexbuczynsky, 2021-02-22:
That's a good point. It seems the type declaration missing is a problem when `tsc` is trying to parse the type declarations.

I believe the original design choice was to make the dependency on `serialport` optional if you are not using modbus RTU (correct me if I am wrong @stefanpoeter). Given the dynamic loading of vanilla javascript, this is not a problem since the module is dynamically loaded in if the RTU files are referenced at runtime. However with typescript, the entire projects types are checked and if the serial port dependency is missing the type info, the transpiler will break. Given this, if we make serialport a required dependency this would fix the problem.

alexbuczynsky, 2021-02-22:
@vn7n24fzkq if you install `serialport` in your project without the type declarations does the tsc build pass or do you need to install the dev dependency on `@types/serialport` as well?

vn7n24fzkq, 2021-02-23:
@alexbuczynsky No, The tsc build would not pass if I don't install `type declarations` in dev dependencies.

stefanpoeter, 2021-02-23:
> I believe the original design choice was to make the dependency on serialport optional if you are not using modbus RTU (correct me if I am wrong @stefanpoeter).

That is absolutely correct. But you could build an interface for the client handling the events for both the net module and the serialport module. You can then make wrapper for both modules. Shouldn't be that hard. Correct me if I am wrong @alexbuczynsky

vn7n24fzkq, 2021-02-24:
@stefanpoeter Thanks for reply.
Yes, I can make a wrapper for both modules, but I am not using modbus RTU.

My temporary solve way is install `serialport` and `@types/serialport` to avoid `tsc` build error.
It is a little weird that I need install a module I don't need. 

Like @alexbuczynsky said,  vanilla javascript use dynamic loading, so we can choice which module is not necessary in vanilla javascript case.

If we don't want to add `net` and `serialport`  module into required dependency then maybe we can add a doc to remind that typescript need install both modules for type declaration when `tsc` build? ( Not sure would other js compilers receive same error )

alexbuczynsky, 2021-02-24:
I like the interface approach. They should honor the abstraction of the specific function / properties we need from these two libraries (`net` and `serialport`).

I will take a look at how to design the interface to work with `tsc`.

alexbuczynsky, 2021-02-24:
Upon further investigation, it seems like in Typescript 3.8 they [introduced importing just the type information](https://www.typescriptlang.org/docs/handbook/release-notes/typescript-3-8.html#type-only-imports-and-export) from a module. Since this library only needs to access the type information and not the classes themselves, this might be a nice approach.

vn7n24fzkq, 2021-02-25:
@alexbuczynsky Thanks a lot!
That would be a good approach. 
I will try it tomorrow.

vn7n24fzkq, 2021-02-26:
I can successfully build by installed `@type/serial` and don't need to install `serialport` now.

stefanpoeter, 2021-02-26:
I think that resolves this issue. Good work :-)

closed 2021-02-26 by stefanpoeter

yorch, 2021-07-21:
@stefanpoeter just faced this issue and found this discussion, thanks for all the good work!

Just my 2 cents, I think the correct approach given that `jsmodbus` package type definition includes `serialport` ts definition, would be to include `@types/serialport` as `dependency`. I don't think TS / Node communities has solved the problem in any other more efficient way unfortunately, but it's what other libraries / packages have to do.

https://stackoverflow.com/a/46011417/443600

tukusejssirs, 2021-12-01:
My two cents: In TS in Nest.js (but Nest.js seems irrelevant), after installing `jsmodbus`, and starting the app, it would fail to start because it does not find `@types/serialbus`. After installing that package manually, it works as expected.

I think that `@types/serialbus` should be a dependency of `jsmodbus`, not of the program that uses `jsmodbus`.