Cloud-Automation/node-modbus #281: how to read registers from meter?

dyaacov, 2021-03-22:
Hi Guys,

I have a meter and a modbus server (I used some old npm library).

The server is listening to incoming connections from the meter and upon an incoming connection, I call connection.readInputRegisters..

How can I implement the same with this library?

Thanks
Dekel

stefanpoeter, 2021-03-24:
I guess you want to read data from the meter? Isn't that supposed to be the server? Or is you meter connecting to a modbus server and therefor acts like a client?

dyaacov, 2021-03-24:
all the meters act like clients and connected to a modbus server.
the meres initiate the connection every 1 minutes and when a connection is
open, the server asks for the registered on the same open connection. (This
way it is easy to communicate between the meters and a remote server
located in cloud)

stefanpoeter, 2021-03-24:
Ok, this way you simply need to start a modbus server and setup the buffers for the input registers. Data will then be stored in the buffers and you can read it at any given time. If you want to intercept or handle the writeInputRegisters Requests there are pre and post handler.

dyaacov, 2021-03-24:
how do I implement the 'on connection received' event in the server?

`
const modbus = require('jsmodbus')
const net = require('net')
const netServer = new net.Server()
const server = new modbus.server.TCP(netServer)

netServer.listen(502)
`

stefanpoeter, 2021-03-24:
Use the sockets 'connection' event

https://nodejs.org/api/net.html#net_event_connection

dyaacov, 2021-03-24:
Someone told me that the first connection is to wake up the meter and only then you can read the values.

the following code throws

`
UserRequestError {
  err: 'Offline',
  message: 'no connection to modbus server',
  response: undefined
}
`

netServer.on('connection', function (socket) {
  console.log('netServer incoming connection')
  const client = new Modbus.client.TCP(socket, 'unitId')
  
  client.readCoils(0, 13).then(function (res) {
    console.log('readCoils', res)
  }).catch(e => console.error(e))
})

netServer.listen(502)

alexbuczynsky, 2021-03-25:
What kind of meter are you using? Every modbus meter I have ever used acts as its own server and you connect to it over port 502 as a client.

closed 2021-05-19 by stefanpoeter

dyaacov, 2021-08-01:
I would like to reopen the discussion.
We are using SATEC meter.

The way it works, the meter open connection to the server and then the server request for holding registers.
this way, you don't need to know the meter ip and the meter can be configured with DHCP.