Cloud-Automation/node-modbus #290: rtu client, write single coil response crc does not match

layliaiyong, 2021-05-29:
debug code
```
  modbus-client received data +16ms
  rtu-response-handler receiving new data +6ms
  rtu-response-handler buffer <Buffer 03 05 00 00 ff 00 8d d8> +0ms
  rtu-response-handler clone buffer <Buffer 03 05 00 00 ff 00 8d d8> +1ms
  rtu-response address 3 buffer <Buffer 03 05 00 00 ff 00 8d d8> +16ms
  response-factory fc 5 payload <Buffer 05 00 00 ff 00 8d d8> +15ms
  rtu-response-handler crc 55437 +3ms
  rtu-response-handler reset buffer from 8 to 0 +0ms
  rtu-response-handler clone buffer <Buffer > +1ms
  rtu-response-handler not enough data available to parse +1ms
  rtu-client-request-handler new response coming in +55ms
  rtu-client-request-handler create crc from response <Buffer 03 05 00 00 00 00> +1ms
  rtu-client-request-handler CRC does not match 55437 !== 10444 +1ms
```
buffer `03 05 00 00 ff 00`  has been changed to `03 05 00 00 00 00`

layliaiyong, 2021-05-29:
@/src/response/write-single-coil.ts line 57
```
this._value = value === 0xFF00 ? 0xFF00 : 0x0000
```
modify
```
this._value = value === 0xFF00 || value === true ? 0xFF00 : 0x0000
```

stefanpoeter, 2022-01-13:
Merged it a while ago. Good work.

closed 2022-01-13 by stefanpoeter