Cloud-Automation/node-modbus #295: modbus.server.RTU responds on all addresses

digitalcraft, 2021-08-15:
Hi,

Am I missing something here?
modbus.server.RTU responds on all addresses.

The constructor takes not ability to set the slave address.
I can't see any logic for conditionally responding if the address is matched.
The address is actually pulled off from the RequestFactory.

RequestFactory.fromBuffer(buffer.slice(1))

How can this be used on a serial network to create a slave where more than 1 slave exists?

stefanpoeter, 2021-08-16:
Hey @digitalcraft, what version are you using?

digitalcraft, 2021-08-16:
Hey Stefan,

I pulled from master.
If there is some better way to use the module. Could you create something
similar to simpleserver.js under the serial folder?

It should be possible to set the slave address, so that it only responds to
packets that match it's address.

stefanpoeter, 2021-08-16:
Hey,

the current development branch is v4.0-dev. I am not sure what is currently on the master branch.

digitalcraft, 2021-08-16:
I think thats the same branch.
So it might be worth while adding an example Serial server then?

stefanpoeter, 2021-08-16:
The way I see it, the packages for other devices can be trashed. It seems as no one else ever used this package as a server via RTU. @digitalcraft  do you care to make a patch?

digitalcraft, 2021-08-16:
Thats what I was thinking, just confirming.
I will make a quick patch, which you can adjust.