Cloud-Automation/node-modbus #296: Handle modbus-tcp messages with too short payload

arobal, 2021-08-24:
If a device fails to send a correct (complete) modbus-tcp response, this currently results in a timeout (as message is too short to be handled).
Because the buffer of the ModbusTCPClientResponseHandler will not be resettet/reduced in this case, following valid messages can't be processed any more but fail with further timeout-errors.

[image]
[image]

This problem can be solved by manually resetting the buffer or do a full reconnect, but is there any better/proposed way to handle this? (dropping the invalid message but beeing able to continue reading)

Thank you

arobal, 2021-08-24:
Here a screenshot of the (unprofessional workaround) solution for beeing able to further communicate with the Modbus device:
[image]

stefanpoeter, 2021-08-24:
If the device fails to send a correct message than the software on the device should be fixed. You can of course fork this package and apply your changes and work from here on.

closed 2022-01-13 by stefanpoeter