Cloud-Automation/node-modbus #302: Address 0 cannot be read

martijndierckx, 2022-01-12:
Any idea why I am unable to read address 0 of the input registers?
When I try to do so on a SmartEVSE, for which I know address 0 exists:
```await conn.readInputRegisters(0, 1);```
I only get:
```
  modbus-client issuing new read input registers request +0ms
  tcp-client-request-handler registrating new request transaction id 1 unit id 1 length 5 +0ms
  user-request creating new user request with timeout 3000 +0ms
  client-request-handler flushing +0ms
  client-request-handler flushing new request <Buffer 00 01 00 00 00 06 01 04 00 00 00 01> +0ms
  client-request-handler request fully flushed, ( error: undefined ) +1ms
``
A response never arrives, and I do not get a timeout error.
If I try an unexisting address, I do get a timeout.

While, if I try to read address 1 of the input registers, everything works perfectly fine:
```await conn.readInputRegisters(1, 1);```
results in:
```
  modbus-client issuing new read input registers request +0ms
  tcp-client-request-handler registrating new request transaction id 1 unit id 1 length 5 +0ms
  user-request creating new user request with timeout 3000 +0ms
  client-request-handler flushing +0ms
  client-request-handler flushing new request <Buffer 00 01 00 00 00 06 01 04 00 02 00 01> +0ms
  client-request-handler request fully flushed, ( error: undefined ) +1ms
  modbus-client received data +95ms
  tcp-response-handler receiving new data <Buffer 00 01 00 00 00 05 01 04 02 00 f0> +0ms
  tcp-response-handler buffer <Buffer 00 01 00 00 00 05 01 04 02 00 f0> +0ms
  tcp-response tcp header complete, id 1 protocol 0 length 5 unitId 1 +0ms
  tcp-response buffer <Buffer 00 01 00 00 00 05 01 04 02 00 f0> +1ms
  response-factory fc 4 payload <Buffer 04 02 00 f0> +0ms
  tcp-response buffer contains a valid response body +0ms
  tcp-response-handler response id 1 protocol 0 length 5 unit 1 +1ms
  tcp-response-handler reset buffer from 11 to 0 +0ms
  tcp-response not enough data available +0ms
  tcp-response-handler not enough data available to parse +0ms
  client-request-handler incoming response +94ms
  client-request-handler resolving request +0ms
  user-request request completed in 95 ms (sat in cue 0 ms) +95ms
  client-request-handler flushing +0ms
  client-request-handler no request to be executed +0ms
```

stefanpoeter, 2022-01-13:
@martijndierckx A working example would be great.

martijndierckx, 2022-01-14:
@stefanpoeter  I'm not sure what you need from me?

stefanpoeter, 2022-01-14:
Create a modbus server and read address zero from there. Publish it here so I can reproduce the error.

closed 2022-04-20 by stefanpoeter