Cloud-Automation/node-modbus #304: request fc and response fc does not match.

anesteziker, 2022-02-28:
Hello, when I use the single register write method, I get the following error after the first few submissions. I need your help on this.

[Arguments] {
  '0': UserRequestError {
    err: 'OutOfSync',
    message: 'request fc and response fc does not match.',
    response: undefined
  }
}

anesteziker, 2022-02-28:
[Arguments] {
  '0': UserRequestError {
    err: 'OutOfSync',
    message: 'request fc and response fc does not match.',
    response: undefined
  }
}
[Arguments] {
  '0': UserRequestError {
    err: 'OutOfSync',
    message: 'request fc and response fc does not match.',
    response: undefined
  }
}

stefanpoeter, 2022-03-21:
Can you provide some code to reproduce this?

closed 2022-04-20 by stefanpoeter