Cloud-Automation/node-modbus #307: RTU server examples.

JaagupAverin, 2022-04-12:
Hello. I am confused about setting up the RTU server. Examples only exist for TCP.
Based on the TCP examples, the following attempt could be concocted, but it does not work - no connection event is generated. It is also unclear how I would specify the node addres/unit ID of this server? Thank you.

```
const modbus = require('jsmodbus')
const SerialPort = require('serialport')
const options = {
    baudRate: 9600
}
const socket = new SerialPort("/dev/ttyUSB0", options)
const server = new modbus.server.RTU(socket)
server.on('connection', function (client) {
    console.log(client);
});
```

stefanpoeter, 2022-04-20:
Hey @JaagupAverin ,

there isn't any. The serve does not check the unit ID. That is clearly a missing feature but it should not be that hard to implement. Would you care to setup a pull request?

KamaxSuen, 2023-01-12:
Hi 

I make a simple patch to address this issue. It work for me

`diff -ru node_modules/jsmodbus/dist/modbus-rtu-server.js patch/jsmodbus/dist/modbus-rtu-server.js
--- node_modules/jsmodbus/dist/modbus-rtu-server.js     2022-11-18 00:39:01.259648905 +0800
+++ patch/jsmodbus/dist/modbus-rtu-server.js    2022-11-18 00:43:13.393902461 +0800
@@ -8,12 +8,12 @@
 const rtu_request_js_1 = __importDefault(require("./rtu-request.js"));
 const rtu_response_js_1 = __importDefault(require("./rtu-response.js"));
 class ModbusRTUServer extends modbus_server_js_1.default {
-    constructor(socket, options) {
+    constructor(socket, options, address) {
         super(options);
         this._socket = socket;
         const fromBuffer = rtu_request_js_1.default.fromBuffer;
         const fromRequest = rtu_response_js_1.default.fromRequest;
-        const client = new modbus_server_client_js_1.default(this, socket, fromBuffer, fromRequest);
+        const client = new modbus_server_client_js_1.default(this, socket, fromBuffer, fromRequest, address ?? 1);
         this.emit('connection', client);
     }
 }
diff -ru node_modules/jsmodbus/dist/modbus-server-client.js patch/jsmodbus/dist/modbus-server-client.js
--- node_modules/jsmodbus/dist/modbus-server-client.js  2022-11-18 00:39:01.263648942 +0800
+++ patch/jsmodbus/dist/modbus-server-client.js 2022-11-18 00:42:41.125622610 +0800
@@ -8,11 +8,12 @@
 const modbus_server_request_handler_js_1 = __importDefault(require("./modbus-server-request-handler.js"));
 const modbus_server_response_handler_js_1 = __importDefault(require("./modbus-server-response-handler.js"));
 class ModbusServerClient {
-    constructor(server, socket, fromBufferMethod, fromRequestMethod) {
+    constructor(server, socket, fromBufferMethod, fromRequestMethod, address) {
         this._server = server;
         this._socket = socket;
         this._requestHandler = new modbus_server_request_handler_js_1.default(fromBufferMethod);
         this._responseHandler = new modbus_server_response_handler_js_1.default(this._server, fromRequestMethod);
+        this._address = address ?? null;
         this._socket.on('data', this._onData.bind(this));
     }
     get socket() {
@@ -30,6 +31,10 @@
                 debug('no request to process');
                 break;
             }
+            if (this._address && request.address != this._address) {
+                debug('request address not match');
+                break;
+            }
             this._responseHandler.handle(request, (response) => {
                 this._socket.write(response, () => {
                     debug('response flushed', response);`

stefanpoeter, 2023-01-12:
Hi @KamaxSuen 

thanks for your work. Can you put this in a pull request.

KamaxSuen, 2023-01-13:
> Hi @KamaxSuen
> 
> thanks for your work. Can you put this in a pull request.

Hi @stefanpoeter 

Thanks for appreciation
I'm newcomer in GitHub. I'm not familiar how to create pull request but I will try

closed 2023-05-19 by stefanpoeter