Cloud-Automation/node-modbus #311: Client IP

Zeppe79, 2022-05-27:
This library is awesome. Is there a way to get the client IP address in a Modbus TCP Server code ? 

`
const net = require('net')
const modbus = require('jsmodbus')
const netServer = new net.Server();
const holding = Buffer.alloc(10000);
const server = new modbus.server.TCP(netServer, {
	holding: holding
})

server.on('connection', function(client){
	console.log('New connection')
	//console.log(client)
})`

In the client object can't find anything suitable 

Thanks

stefanpoeter, 2022-05-29:
The way I see it the client object should contain a socket object from the net module. From there you can call address().

Zeppe79, 2022-06-06:
Thanks for the suggestions. I've solved it with socket.remoteAddress property !

closed 2022-06-06 by stefanpoeter

def-fun, 2023-11-02:
> Thanks for the suggestions. I've solved it with socket.remoteAddress property !

I can get IP in `connection` also:
```
server.on('connection', function(client){
	console.log('New connection')
	console.log(client.socket.remoteAddress)  //IP
})
```

Is it possible to get IP in `postWriteMultipleRegisters` ?
```
server.on('postWriteMultipleRegisters', function (value) {
    console.log('Write multiple registers - Complete: ', holding.readUInt16BE(0))
    // how to get client IP here ?
})
```

stefanpoeter, 2023-11-02:
That would not make much sense since the remote address does not change with every request and is connection specific.