Cloud-Automation/node-modbus #319: ECONNRESET when client disconnect

yususs, 2022-12-14:
How to prevent Tcp Server to exit process when client disconnect?

stefanpoeter, 2022-12-15:
Do you have some example code?

yususs, 2022-12-15:
Yeah sorry about that.

`const modbus = require('jsmodbus')
const net = require('net')
const netServer = new net.Server()
const holding = Buffer.alloc(10000)
const server = new modbus.server.TCP(netServer, {
    holding: holding
})

server.on('connection', function (client) {
  console.log('New Connection')
  console.log(client.socket.remoteAddress);
})

server.on('preWriteSingleRegister', function (value) {
  let address = value.body.address
  let values = value.body.value
  console.log(address, values)
})

server.on('postWriteMultipleRegisters', function (value) {
  let address = value.body.address
  let values = value.body.valuesAsArray
  console.log(address, values)
})

server.on('error', function (value) {
  console.log(value)
})

netServer.listen(502)`

stefanpoeter, 2022-12-16:
Can you turn on debugging (DEBUG=*) and show me what is happening?

closed 2023-05-19 by stefanpoeter