Cloud-Automation/node-modbus #328: client.writeMultipleRegisters(143, [1]) -> err: 'Timeout',

Foxylabstory, 2023-09-05:
Good day, I hope you are going well!
I've meet situation where I'm trying to write multiple registers, registers are change, but I can't wait any answer. I have no guess why, could you explain me please, what needs to do?
`client.writeMultipleRegisters(143, [1])
      .then((resp) => {
        console.log(resp);
      })
      .catch((err) => {
        console.log(err);
      });`
  It command returns: 
  UserRequestError {
  err: 'Timeout',
  message: 'Req timed out',
  response: undefined
}
If I'm doing manually, there is no problem
[image]

And you should to know, client.readHoldingRegisters(4096, 100).then().catch is working perfectly!

Foxylabstory, 2023-09-05:
Also I've tried this issue:
```let b = Buffer.allocUnsafe(2);
 b.writeUInt16BE(1);
client.writeMultipleRegisters(143, b)```
It also doesn't work.

stefanpoeter, 2023-09-05:
Could be anything. Can you provide a reproducable piece of code.

Foxylabstory, 2023-09-06:
Here block that you've asked
```
const Modbus = require('jsmodbus');
const net = require('net');
const socket = new net.Socket();
const client = new Modbus.client.TCP(socket, 206);
const options = {
  'host': '217.118.178.146', // device ip real, wait for your actions
  'port': 4001,
  'autoReconnect': true,
  'logEnabled': true
};

socket.connect(options);

socket.on('connect', function () {
  console.log('socketconnected');
  
  setTimeout(() => {
    console.log(client.connectionState);
    
    client.readHoldingRegisters(11, 3) // 11 begining of reading, 3 number of reading. Working perfectly, thanks a lot.
      .then((resp) => {
        console.log(resp.response._body._values[0]); // expected slaveID of device - 206
        console.log(resp.response._body._values[1]); // expected serial number of device - 10998
        console.log(resp.response._body._values[2].toString(16)); // expected firmware of device - "1223"
      })
      .catch((err) => {
        console.log(err);
      });
  }, 500);
  
  setTimeout(() => {
    console.log(client.connectionState);
    client.writeMultipleRegisters(129, [1]) // should be between 1 and 65535
      .then((resp) => {
        console.log(resp);
      })
      .catch((err) => {
        console.log(err); // every attempt returns -> "err": "Timeout", "message": "Req timed out"
      });
  }, 1000);
})

socket.on('error', function () {
  console.log('socketerror');
})

socket.on('timeout', () => {
  console.log('timeout');
})

socket.on('close', () => {
  console.log('socketclose');
  launchIntervalConnect();
});

socket.on('end', () => {
  console.log('socketend');
});
```

stefanpoeter, 2023-09-06:
Don't see anything wrong, can you provide a tcpdump?

Foxylabstory, 2023-09-07:
Thanks for your attention, problem was in module.exports, I've set new unit id via calling a function it went wrong because in imported modules was previous unit id.
Reading tcpdump helped, we've found that answer contain wrong unit id.
Solved with adding one more method, it will a garantie, that I'll receive updated instance of class.
```
function getClient() {
  return client;
}  
```

closed 2023-09-07 by Foxylabstory

dyaacov, 2024-03-12:
@Foxylabstory 
Can you please provide a full example?
I have some devices that I also get timeout while in other it works fine

dyaacov, 2024-03-12:
The setTimeout(() => { 1000 
to write fixed my issues !!!

Thanks!