Cloud-Automation/node-modbus #330: bug, readHoldingRegisters cannot read over 128

pmingkr, 2023-10-25:
```ts
      const { response } = await client.readHoldingRegisters(
        0,
        129
      );
      console.assert(response.body.valuesAsArray.length === 1);
```

stefanpoeter, 2023-10-25:
That is not very much you are giving me here.

pmingkr, 2023-10-26:
I made sample code for it.
but I met a different error message.
https://codesandbox.io/p/sandbox/hardcore-chihiro-sclvqt?file=%2Findex.js%3A10%2C4

stefanpoeter, 2023-10-29:
The "ERR_OUT_OF_RANGE" probably occurs because you did not provider a buffer for the modbus registers.

pmingkr, 2023-10-30:
> The "ERR_OUT_OF_RANGE" probably occurs because you did not provider a buffer for the modbus registers.

No it's not.
RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be >= 0 and <= 255. Received 258
It happened because it tried to access index 258, but I had requested indices from 0 to 128.

stefanpoeter, 2023-10-30:
I see. The modbus specification allows a total quantity of 125 registers to be fetched with one request. The Error Message comes from the array or buffer that cannot fit more that 255 bytes in there. The error message is a bit cryptic I admit.

[image]

pmingkr, 2023-10-31:
> I see. The modbus specification allows a total quantity of 125 registers to be fetched with one request. The Error Message comes from the array or buffer that cannot fit more that 255 bytes in there. The error message is a bit cryptic I admit.
> 
> [image]

Could you tell me the source of it?
I checked the Modbus Organization's document, but I could not find it.
https://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf

stefanpoeter, 2023-10-31:
You find the PDF here: https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf

pmingkr, 2023-10-31:
> You find the PDF here: https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf

Thanks for your explanation.

jsmodbus can read 127 registers.
it seems it's a also broken behavior.

stefanpoeter, 2023-11-27:
you are right @pmingkr There should be a error when issuing a request that reads more than 127 register. I am always happy for contributions :-)