Cloud-Automation/node-modbus #331: Jsmodbus? Modbus TCP Server Errors?

maufc, 2023-10-25:
Hi.
I'm a little confused about the name of this repo and I found a couple of errors in TCP Server maybe already overcome.
I am using node 18.x and had install jsmodbus, the library I though this repo was of.
This repo is refered as jsmodbus in npm web: 
https://www.npmjs.com/package/jsmodbus?activeTab=readme
But it is name as node-modbus.
When I try to install node-modbus, it says it is only compatible with node 6.x and 8.x so I couldn't try my script with it.
Something about the error found is described here: 
https://github.com/Cloud-Automation/node-modbus/issues/230
readHoldingRegisters can't be called but prereadHoldingRegisters and postreadHoldingRegisters can.
When using postwritemultipleregister, it seems like it executes twice as not matter where you make the request from, a console message shows twice.
I thing not make people uses the server option, maybe this simple errors not a big issue.
Thanks so much!
Regards!

My code is as simple as this: 
```const net = require('net')
const modbus = require('jsmodbus')
const netServer = new net.Server()
const holding = Buffer.alloc(80)
const server = new modbus.server.TCP(netServer, {
  holding: holding
})

server.on('connection', function (client) {
  console.log('New Connection')
})

server.on('postReadHoldingRegisters', function (request, response) {
  console.log("Sent... "+server.holding.toString("hex"));

})

server.on('postWriteMultipleRegisters', function (value) {
  console.log('Registers Wrote!"))
})

netServer.listen(502, ()=> {
   console.log('Modbus Server Started');
});
```
When trying to WriteMultipleRegisters, It show 'Registers wrote' twice.

stefanpoeter, 2023-10-29:
Hey Claus,

the name of the npm and the name of the github repo a two different things. But I agree that this could have been the same name. It's some time ago now and I don't really know any more how it came to this.

I don't have any metrics on how many people use the server but my guess would be that the client is more in the focus. I remember the issue you mentioned, unfortunately there has never been a pull request for this issue. I am happy if you contribute a solution.

maufc, 2023-11-04:
Hi.
Thanks for your answer.
Let me see what can I do to propose any improvement.
Regards!

closed 2023-11-04 by maufc