Cloud-Automation/node-modbus #332: TypeScript Example

cycl0ne, 2023-10-27:
Hi,
first the example using: 
import Modbus from '../../..'

yes from your view ok, but i installed jsmodbus and then this wont work :-P

second: I just copy the example code and execute it:
```
import Modbus from 'jsmodbus'
import { Socket, SocketConnectOpts } from 'net'

const socket = new Socket()

const options: SocketConnectOpts = {
  host: '127.0.0.1',
  port: 8502
}
const client = new Modbus.client.TCP(socket)

const readStart = 0;
const readCount = 10;

socket.on('connect', function () {
  client.readHoldingRegisters(readStart, readCount)
    .then(({ metrics, request, response }) => {
      console.log('Transfer Time: ' + metrics.transferTime)
      console.log('Response Body Payload: ' + response.body.valuesAsArray)
      console.log('Response Body Payload As Buffer: ' + response.body.valuesAsBuffer)
    })
    .catch(handleErrors)
    .finally(() => socket.end())
})

socket.on('error', console.error)
socket.connect(options)

function handleErrors(err: any) {
  if (Modbus.errors.isUserRequestError(err)) {
    switch (err.err) {
      case 'OutOfSync':
      case 'Protocol':
      case 'Timeout':
      case 'ManuallyCleared':
      case 'ModbusException':
      case 'Offline':
      case 'crcMismatch':
        console.log('Error Message: ' + err.message, 'Error' + 'Modbus Error Type: ' + err.err)
        break;
    }

  } else if (Modbus.errors.isInternalException(err)) {
    console.log('Error Message: ' + err.message, 'Error' + 'Error Name: ' + err.name, err.stack)
  } else {
    console.log('Unknown Error', err);
  }
}
```

i get the following error:

const client = new jsmodbus_1.default.client.TCP(socket);
                                    ^
TypeError: Cannot read property 'client' of undefined
    at Object.<anonymous> (/home/pi/Python/_node/dist/app.js:13:39)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47

Any idea?

stefanpoeter, 2023-10-29:
Hey @cycl0ne,

I am always happy for contributions if you want to correct the example.

There error looks like more of a setup problem. Seems like there is something missing when you build the javascript files. Don't see anything wrong with the code.

cycl0ne, 2023-11-22:
Hmm, did a complete clean install. same error. "TypeError: Cannot read property 'client' of undefined"

Understanding Question here:  I just use the example without a server on 127.0.0.1 maybe because of this, since he cant find a connection there?
Shouldnt there be a sort thrown error for this?

stefanpoeter, 2023-11-27:
did you checked this `const client = new jsmodbus_1.default.client.TCP(socket);` and `const client = new Modbus.client.TCP(socket)`

What is happening there?

cycl0ne, 2023-11-27:
in the latest try above.. i get the error that client is undefined. but mouseover in IDE (Code-Server) says: its a const.. but executing it after ts compilation throws: undefined.

i will check the first example i used.

cycl0ne, 2023-12-02:
hmm trying to locate still the error.. using your cli tool: jsmodbus, but now i get the following error:

jsmodbus fc04 192.168.1.13 225 1285

Attempting to connect to 192.168.1.13:502 at unitId 225 ...
Connected
[Arguments] {
  '0': UserRequestError {
    err: 'ModbusException',
    message: 'A Modbus Exception Occurred - See Response Body',
    response: ModbusTCPResponse {
      _id: 1,
      _protocol: 0,
      _bodyLength: 3,
      _unitId: 225,
      _body: ExceptionResponseBody { _fc: 4, _code: 3 }
    }
  }
}

with qModbus (tool for windows),Loxone SmartHome and python (with the modbus module) i can read the values with this setting: ip/unit/adress.

i think i will give up on nodejs and go back to python. 🤣

stefanpoeter, 2023-12-02:
Hey @cycl0ne ,

the specification states the following regarding error code 4

[image]

cycl0ne, 2023-12-02:
yes, but here again: i just use the CLI Tool: 
jsmodbus fc04 192.168.1.13 100 844

Output in Bash:
Attempting to connect to 192.168.1.13:502 at unitId 100 ...
Connected
[Arguments] {
  '0': UserRequestError {
    err: 'ModbusException',
    message: 'A Modbus Exception Occurred - See Response Body',
    response: ModbusTCPResponse {
      _id: 1,
      _protocol: 0,
      _bodyLength: 3,
      _unitId: 100,
      _body: ExceptionResponseBody { _fc: 4, _code: 3 }
    }
  }
}

here the result with qmodmaster tool:
[image]

so i dont modify query data or anything else.. just using the cli tool.

stefanpoeter, 2023-12-02:
Don't know what you expect, the bin/jsmodbus.js defines the parameters for a fc04 call like this

 ```.command('fc04 <host> <unitId> <range>')```

I think you need to change your command to something like

```jsmodbus fc04 192.168.1.13 unitid startRegister endRegister```

closed 2023-12-02 by cycl0ne