Cloud-Automation/node-modbus #342: Build using Typescript error: Cannot use namespace 'SerialPort' as a type.

Zandor300, 2024-07-02:
Receiving the following error if I want to build using Typescript. Before I was just running the plain Javascript.

```
jsmodbus:           4.0.6 (Since that is the last one published on npm)
serialport:        12.0.0
@types/serialport:  8.0.5
typescript:         5.5.3
```

```
$ tsc
node_modules/jsmodbus/dist/modbus-rtu-client.d.ts:6:55 - error TS2709: Cannot use namespace 'SerialPort' as a type.

6 export default class ModbusRTUClient extends MBClient<SerialPort, ModbusRTURequest> {
                                                        ~~~~~~~~~~

node_modules/jsmodbus/dist/modbus-rtu-client.d.ts:9:25 - error TS2709: Cannot use namespace 'SerialPort' as a type.

9     constructor(socket: SerialPort, address: number, timeout?: number);
                          ~~~~~~~~~~

node_modules/jsmodbus/dist/modbus-rtu-server.d.ts:6:25 - error TS2709: Cannot use namespace 'SerialPort' as a type.

6     constructor(socket: SerialPort, options?: Partial<IModbusServerOptions>);
                          ~~~~~~~~~~

node_modules/jsmodbus/dist/rtu-client-request-handler.d.ts:7:83 - error TS2709: Cannot use namespace 'SerialSocket' as a type.

7 export default class ModbusRTUClientRequestHandler extends MBClientRequestHandler<SerialSocket, ModbusRTURequest> {
                                                                                    ~~~~~~~~~~~~

node_modules/jsmodbus/dist/rtu-client-request-handler.d.ts:11:25 - error TS2709: Cannot use namespace 'SerialSocket' as a type.

11     constructor(socket: SerialSocket, address: number, timeout?: number);
                           ~~~~~~~~~~~~

Found 5 errors in 3 files.

```

stefanpoeter, 2024-07-02:
The module is build against serialport version `"serialport": "^7.1.5"` see package.json.

Zandor300, 2024-07-08:
@stefanpoeter I use it with serialport v12 and it works fine, only the exact types used are incorrect when building using Typescript.

stefanpoeter, 2024-07-11:
That sounds like incompatibilties. If you want to port I am happy for any PR. As said it is compatible to 7.1.5.

swisspol, 2024-08-15:
This seems to work around the compiling issue in TypeScript:
```
npm install serialport@9
npm i --save-dev @types/serialport
```

stefanpoeter, 2024-09-12:
Have you checked adding this to the package.json?

Oniokey, 2024-12-14:
> This seems to work around the compiling issue in TypeScript:
> 
> ```
> npm install serialport@9
> npm i --save-dev @types/serialport
> ```

Both the `@9` version and the `latest` version of `SerialPort` raise the same issue

closed 2025-01-07 by stefanpoeter