Cloud-Automation/node-modbus #355: Missing `commander` dependency

joshkel, 2025-09-26:
I looked into using the jsmodbus [CLI](https://www.npmjs.com/package/jsmodbus#cli) but encountered an error:

```
  .command('fc01 <host> <unitId> <range>')
   ^

TypeError: program.command is not a function
```

It appears that this is because jsmodbus lists its commander@^3.0.2 dependency under devDependencies, so it's not installed at runtime, and the package in my project instead gets my project's commander@^12.1.0, which isn't compatible.

I started looking into opening a PR for this, but several of jsmodbus's dependencies are out of date. (I couldn't install packages because the Node.js APIs have changed for serialport, and updating serialport exposed issues with an out-of-date TypeScript, and updating TypeScript apparently exposed issues with tslint.) If you're interested in a PR that updates several packages, I can work on one, but I didn't want to make larger changes without checking with you.

Thank you.

stefanpoeter, 2025-09-28:
Hi @joshkel ,

updating the dependencies sounds great. Something I had on my bucket list for some time now. I would be more than happy if you want to open up a PR. For compatibility reasons I would make that version 5.0.0.

Barneey96, 2026-09-10:
Hello @stefanpoeter , @joshkel , 

I opened a general PR for node 20+ support: https://github.com/Cloud-Automation/node-modbus/pull/360 which would also fix this.